dresden-elektronik/deconz-rest-plugin #682: Sengled light bulb showing as end device

xhemp, 2018-07-09:
I have just bought a sengled zigbee light bulb to test and was able to pair with deCONZ just fine but it's being recognised as an end device instead of a gateway.
Is there anything I can grab from deCONZ so you can have a look?

[image]

manup, 2018-07-09:
Screenshot of the NodeInfo panel would be interesting, but I'm afraid it is just as it looks and the light might not act as a router (which is unusual but but totally allowed by the spec).

xhemp, 2018-07-09:
I understand.
Here is the screenshot:

[image]

manup, 2018-07-10:
From here it looks like the device is indeed not a router, device type RFD means reduced function device with no routing capabilities. Might be error in the node descriptor but only a sniffer log would tell.

xhemp, 2018-07-10:
@manup Thanks for having a look at it, I had emailed Sengled and they confirmed it's an RFD, I am closing this one.

closed 2018-07-10 by xhemp