dresden-elektronik/deconz-rest-plugin #684: Inconsistent web socket handshake

Kane610, 2018-07-10:
Hi @manup 

HASS users have identified an inconsistent behavior in the web socket handshake. It seems to be dependent on host package versions that deCONZ depends on. For our community deCONZ docker container it seems to always work, but for different instances of deCONZ running directly on host it can be a hit or miss. Meaning that one system might not work with the home assistant implementation. Is this something you are aware of / can look at?

manup, 2018-07-10:
We use the Qt5 Websocket module which implements everything like handshake.
There might be issues in older versions of Qt, would be interesting to know which versions of Qt5 have the problems.

With recently updated Raspbian Stretch Qt 5.7 is installed, before it was Qt 5.3.

Kane610, 2018-07-10:
Thanks for the information. I will try to verify if there is a difference in behavior between versions

mathos77, 2018-07-10:
@Kane610 I am running:

```
qt5-default/stable,now 5.7.1+dfsg-3+rpi1 armhf [installed,automatic]
qt5-doc/stable 5.7.1-2 all
qt5-doc-html/stable 5.7.1-2 all
qt5-gtk-platformtheme/stable,now 5.7.1+dfsg-3+rpi1 armhf [installed,automatic]
qt5-image-formats-plugins/stable 5.7.1~20161021-2 armhf
qt5-qmake/stable,now 5.7.1+dfsg-3+rpi1 armhf [installed,automatic]
qt5-qmltooling-plugins/stable 5.7.1-2+rpi1 armhf
qt5-style-plugins/stable 5.0.0+git16.g7aa4764-1 armhf
qt5ct/stable 0.34-1~bpo9+1 armhf
qt5ct-dbgsym/stable 0.34-1~bpo9+1 armhf
qt5gstreamer-dbg/stable 1.2.0-4 armhf
qt5keychain-dev/stable 0.7.0-3 armhf
qt5qevercloud-dev/stable 3.0.3+ds-2 armhf
qt5serialport-examples/stable 5.7.1~20161021-2 armhf
```

manup, 2018-07-10:
Looks like the latest version shipped with Raspbian Stretch.

Kane610, 2018-07-10:
@mathos77 and it works for you now?

mathos77, 2018-07-10:
yes it does work with 0.73.0 and 0.73.1

Kane610, 2018-07-10:
Upgrading my raspberry to stretch to see if that solves the issue, I did indeed have QT 5.3

Kane610, 2018-07-10:
That seems indeed to be the problem. Thanks for the support ❤️

closed 2018-07-10 by Kane610