dresden-elektronik/deconz-rest-plugin #686: Xiaomi button stopped working

svenove, 2018-07-13:
I have 2 Xiaomi buttons, but one of them suddenly stopped working. I've replaced the battery and tried to open the network and pressing the "link"-button. 

It shows up in the REST-API:
`{
  "config": {
    "battery": null,
    "on": true,
    "reachable": false,
    "temperature": 0
  },
  "ep": 1,
  "etag": "94a6f355dceb963e05e961a10d4b8acc",
  "manufacturername": "LUMI",
  "mode": 1,
  "modelid": "lumi.sensor_switch",
  "name": "Switch Xiaomi",
  "state": {
    "buttonevent": 1002,
    "lastupdated": "2018-06-30T07:15:29"
  },
  "type": "ZHASwitch",
  "uniqueid": "00:15:8d:00:01:f4:20:23-01-0006"
}`

But in the GUI, I can't see it.

Running 2.05.32.
Any suggestions on how to proceed?

manup, 2018-07-14:
While the sensor search is running you may try to press the small setup button for 10 seconds.
It should then rejoin, settings are preserved.

svenove, 2018-07-14:
That did the trick! :)

closed 2018-07-14 by svenove