dresden-elektronik/deconz-rest-plugin #688: Hue Dimmer and Xiaomi Aqara Smart SwitchLight Double lost connection after few hours

jantje88, 2018-07-16:
I am using a RaspBee Premium on a RPi3 with HASS and the [deCONZ add-on](https://www.home-assistant.io/components/deconz/) and have 20 IKEA Tradfri lights (GU10 en E27), 3 Osram Smart+ sockets, 10 Xiaomi Aqara sensors (window/door, mult and motion) and a [Xiaomi Aqara Smart Wireless Switch](https://www.gearbest.com/access-control/pp_626695.html) connected which works without any problem for a few weeks now.

However, a [Hue Dimmer](https://www2.meethue.com/nl-nl/p/hue-dimmer/8718696743157) and a [Xiaomi Aqara Smart Light Switch Wireless Version Double](Xiaomi WXKG02LM Aqara Smart Light Switch Wireless Version) lose connection after a few hours. Rebooting HASS (multiple times) does not solve this issue. The switches turn to grey after a while and no events are coming to HASS anymore. After removing both switches and repair them, they will work for a few hours again.

Deconz: 2.05.31
HASS: 0.73.0

manup, 2018-07-16:
A few things to check:
- Which RaspBee firmware is installed? You can see this in: *Phoscon App > Menu > Gateway*.
- To which parent nodes are the problematic devices connected (deCONZ GUI)?

If possible it's best to force these devices to connect to the RaspBee directly or to Philips Hue lights, since these provide the most stable mesh functionality.

jantje88, 2018-07-16:
- RaspBee firmware 26050500 is installed
- I am not sure If I am able to see this when using the HASS Add-on on hass.io? I have acces the the 'Wireless Light' and Phoscon App, but can not find any routing table or something. I have a few Tradfri lights near the RaspBee so it is possible that the Hue Dimmer and Xiaomi switches are connected to them (after a while). How to force these devices to connect to the RaspBee directly (I do not have any Hue lights)?

manup, 2018-07-16:
> RaspBee firmware 26050500 is installed

Oh that's pretty old firmware, please update it via:
1) Create a backup in the Phoscon App
2) Close deCONZ
3) Update Firmware
`$ sudo GCFFlasher_internal -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261f0500.bin.GCF`

> I am not sure If I am able to see this when using the HASS Add-on on hass.io? I have acces the the 'Wireless Light' and Phoscon App, but can not find any routing table or something. I have a few [...]

It is only visible when you see the desktop on the Raspberry Pi where deCONZ GUI is shown, if you run Raspbian as headless mode it is currently not possible to see the GUI.

manup, 2018-07-17:
I've no experience with hass.io setup but maybe the following link is helpful:

https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware

jantje88, 2018-07-17:
Just managed to SSH on port 22222 to the host OS of Hass.io but stuck on step 3 of https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware.
I have created another issue for this: https://github.com/marthoc/docker-deconz/issues/33

jantje88, 2018-07-18:
Solution can be found here: https://community.home-assistant.io/t/marthoc-deconz-addon-cant-connect-lights/57143/12?u=marthocoo