dresden-elektronik/deconz-rest-plugin #689: Hive Smart Plug being recognised as Light

xhemp, 2018-07-18:
Greetings!

Just got a Hive Smart Plug and it looks like it's being recognised as a Light.
Attached are some screenshots from the Cluster Node info:

[image]
[image]

Let me know if i can provide anything else.

xhemp, 2018-07-18:
Cool, thanks for the explanation.
I have taken a screenshot of everything, let me know if there's anything else I can add.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

GadgetUK, 2018-07-30:
I believe the Salus SP600 the same as the Hive Socket so getting support for this would be great. I'll try and post attributes to confirm

closed 2018-12-05 by stale[bot]