dresden-elektronik/deconz-rest-plugin #692: IKEA motionsensor (TRÅDLÖS) gets different unique id:s

helto4real, 2018-07-21:
Home Assistant, home automation software uses some kind of unique id from deconz. For my IKEA motion sensors it sometimes changes.

**Example**

90:fd:9f:ff:fe:71:50:b5-01-0006 vs 90:fd:9f:ff:fe:71:50:b5-01-1000

it seems to be the last 4 digits that are changing. 

When this happens the devices get new names and I cant use them until I reconfigure Home Assistant.

Please check the logic when the id:s change for IKEA motion sensors and try to keep them consistent.

Thanks!

ebaauw, 2018-07-21:
The `uniqueid` is composed of the ZigBee mac address, endpoint, and cluster.  I would expect the last one to be 0006 for the Trådfri motion sensor (corresponding to the _OnOff_ cluster), but mine shows 1000 as well.

Did yours change from 1000 to 0006 or from 0006 to 1000?  Did you re-pair the motion sensor?

helto4real, 2018-07-21:
It changed from 0006 to 1000, I did not re-pair, just restart due to uppgrade of Home Assistant

manup, 2018-07-21:
When loading devices from database the unique it gets recreated, looks like here results are somehow different when on creation of this particular sensor. So we need to check that the selected clusterId will be the same as used on creation.

helto4real, 2018-07-21:
Sounds good. Thanks!

ebaauw, 2018-11-24:
Looks like this hasn't been addressed yet.  Lemme rectify that straight away ;-)

As far as I can tell, apart from this bug, the _ZLL Commissioning_ cluster is only used on for ZHASwitch resources.  The only `/sensors` resources with a `uniqueid` ending with `-1000` in my possession are all IKEA Trådfri devices: a motion sensor (ZHAPresence), two remotes and a dimmer (ZHASwitch).  The logic for determining the cluster seems the same for creating and restoring the ZHASwitch, but the code is different.  Brr.

I think it makes sense to use the (client) _On/Off_ cluster for the ZHASwitches as well (or at least let the _On/Off_ cluster take precedence over the _Commissioning_ cluster).  This would be more consistent with other switches working similarly (i.e. issuing _OnOff_ commands when a button is pressed): ZHA switches that will never have a _ZLL Commissioning_ cluster (e.g. the ubisys D1 high voltage inputs), and ZZL switches, that have the _ZLL Commissioning_ cluster on a different endpoint (e.g. the innr remote), or don't have it (e.g. the Hue dimmer on its ZLL endpoint, but we use the manufacturer specific cluster on the ZHA endpoint for these).

@manup, what do you think?  Changing this might confuse software using the `uniqueid` to access the sensors, instead of the sensor id.

manup, 2018-11-25:
>@manup, what do you think? Changing this might confuse software using the uniqueid to access the sensors, instead of the sensor id.

Yes this is a tricky one I wouldn't change existing `uniqueids`, beside apps also rules can make use of them and the change would break these.

When I remember correctly the *ZLL Commissioning* cluster is used to filter more sharp for specific devices and leave others out which have an On/Off client cluster but should not create a resource for it.

I need to check the code, but I'm more in favor to improve implementation so that existing scheme including `-1000` would be used to keep compatibility  and check correctness of identical `uniqueid` at creation and database loading.

ebaauw, 2018-11-25:
> I'm more in favor to improve implementation so that existing scheme including `-1000` would be used to keep compatibility and check correctness of identical `uniqueid` at creation and database loading.

For sensors, `uniqueid` is actually persisted in the database, in a column of its own.  On database load, we could simply restore the saved value, rather than re-construct it from the fingerprint.  Then, we can correct the logic on creating new sensor resources as changes would only happen when the sensor is deleted and re-paired.

For lights, `uniqueid` is also persisted, in the inaptly named `mac` column (did this change or did I never notice before?), so we can do the same.

manup, 2018-11-25:
It would be super to just load the uniqueid from the database but unfortunately this is more challenging when expected :)

The reason behind the recreation of the `uniqueid` on database load is the ability to correct invalid `uniqueid` from older versions or backups.

Since there is a quite diverse selection of deCONZ versions out there ranging from 2.04.35 upwards the database code must somehow deal with that when the user updates.

Maybe we can figure out a helper method to generate the uniqueid based on fingerprint, which can be used by both sensor creation and database loading code. Could be handy and the method is a good candidate for automatic tests.

manup, 2018-11-25:
Yes something like `bool uniqueIdForSensor(*uniqueid, mac, fingerprint, type [, modelid])`.
Since we already know how the result must look like for existing data sets verification should be easy so we can be sure nothing breaks.

closed 2019-04-01 by stale[bot]