dresden-elektronik/deconz-rest-plugin #695: Hue dimmer and its parent node stops working after a few days

nero01, 2018-07-22:
Hi,
first thank you for creating this great ZigBee module and software bundle with frequent updates. 
I am using the RaspBee module on a RPi3 with HASS and the deCONZ add-on. I have 3x IKEA GUI10 LED spots, 1x E27 IKEA, 2x E27 Philips Hue, 3x Philips Hue Dimmer and 2x Hue motion sensor in use and control everything from HASS.
The issue is related to the connection of the Philips Hue Dimmer to the network, it always stops working after a few hours/days. The remote shows a green dot and short after a red dot, but the last triggered value does not update anymore and HASS does not see the button presses.
This time i checked the parent node and it was a GU10 bulb when the remote first joined the network and when the issue happened again the parent node changed to a Philips Hue E27 bulb. The GU10 bulb that the remote was first connected to, has now completely lost connection and is shown in red in the GUI. 

I have tried to do the following steps to solve the issue:
1. Restart deCONZ
2. Delete node in GUI and via API
3. Put the remote beside the RaspBee module and press and hold the button on the back to reset it
4. I opened the network and the remote joined in again.
5. I checked the parent node and its never the RaspBee module itself but usually one of the connected lights.

I am running the latest beta version of deCONZ and the latest HASS update.

deCONZ: 2.05.32
Firmware: 26210500

nero01, 2018-08-12:
Guys, can anyone help, i tried to downgrade the firmware to 261E0500 but it still happend after about 2 weeks. I am frustrated and do not know what i can do. Currently the hue dimmers are not reliable anymore, they stop working after some time.

manup, 2018-08-12:
Currently the best advice is to force the dimmer to connect to either RaspBee or one of the hue lights. You can force this by not powering the IKEA lights while joining the dimmer (just pressing the buttons might do the trick as well). The main problem is the end-device implementation in the IKEA bulb firmware which removes end devices from the child table if they sleep for some time, it's not a bug but doesn't work well with various ZLL end devices. Hopefully this will be fixed in future IKEA firmware (which we can update).

To downgrade RaspBee please use firmware  0x261f0500 contained in the deCONZ package.

nero01, 2018-08-16:
Thank you very much for your help. I will test that and hope to resolve my issues.

eltoro81, 2018-08-19:
I had similar issues with Hue Motion detector. Twice it has suddenly stopped working including temp/lux reporting. Re-adding seems to work for a little while. I paired it next to the Conbee and it worked fine. Then moving it into position and it stopped within a couple of hours. It is definately meshing through an Ikea Trådfri bulb.

closed 2018-12-24 by stale[bot]