dresden-elektronik/deconz-rest-plugin #696: IKEA TRÅDFRI Remote

DrJDizzle, 2018-07-23:
They work fine with the deconz gateway for dimming and turning on and off.
However, tunable white leds cannot be controlled with the buttons on the remote.
If they are directly paired without the gateway, they are able to change color temperature.

manup, 2018-07-24:
>If they are directly paired without the gateway, they are able to change color temperature.

To my knowledge this works only with IKEA lights and is an extension from IKEA.

ebaauw, 2018-07-24:
Yes, for the left and right buttons, the dimmer issues manufacturer-specific _Scenes_ cluster commands, known only to IKEA lights.

I managed twice somehow to get the buttons to work while both the dimmer and the IKEA light where linked to the deCONZ network.  In both cases the functionality broke spontaneously (?) and I haven’t been able to reproduce this.  I confirmed through sniffing that the commands are still sent by the dimmer and that they are forwarded over the mesh network by non-IKEA lights.  My current theory is that the light contains some magic group and/or scene on factory reset, which are needed to react to the manufacturer specific commands.  Somehow these get broken by deCONZ when updating groups and/or scenes on the light.

manup, 2018-07-24:
When I remember correctly the remote installs scenes and group membership to the (IKEA) lights when it is paired directly to it via the pairing button, this may work even if the remote is already connected to the gateway.

Since the gateway doesn't know these groups/scenes they might get deleted by deCONZ under some conditions.

ebaauw, 2018-07-24:
But then deCONZ should be able to re-create the scenes when an IKEA light is added to the group for the dimmer.  Or do they use manufacturer-specific extentions to the scene as well?

ademobile, 2018-07-26:
I found a rule that had got written at some point, that when I fixed the groups it allows the 'left' and 'right' button to adjust the CT. Examples are:

"1": {
    "actions": [
      {
        "address": "/groups/64020/action",
        "body": {
          "ct_inc": -32,
          "transitiontime": 4
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/3/state/buttonevent",
        "operator": "eq",
        "value": "4002"
      },
      {
        "address": "/sensors/3/state/lastupdated",
        "operator": "dx"
      }
    ],
    "created": "2018-03-17T11:22:05",
    "etag": "792e8a435e6c4c2e5f03cddae8d4af0e",
    "lasttriggered": "none",
    "name": "default-ct-cw-1",
    "owner": "deCONZ",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 0
  }

and 

"2": {
    "actions": [
      {
        "address": "/groups/64020/action",
        "body": {
          "ct_inc": 32,
          "transitiontime": 4
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/3/state/buttonevent",
        "operator": "eq",
        "value": "5002"
      },
      {
        "address": "/sensors/3/state/lastupdated",
        "operator": "dx"
      }
    ],
    "created": "2018-03-17T11:22:05",
    "etag": "792e8a435e6c4c2e5f03cddae8d4af0e",
    "lasttriggered": "none",
    "name": "default-ct-ww-1",
    "owner": "deCONZ",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 0
  }

I am not sure if this is what you are after?

manup, 2018-07-27:
>But then deCONZ should be able to re-create the scenes when an IKEA light is added to the group for the dimmer. Or do they use manufacturer-specific extensions to the scene as well?

Yes I think we should be able to use the same commands to create theses, not sure sure if the scenes are special and require a extension or if they just happen to use fixed scenes numbers.

The next deCONZ version will contain the new switch editor as beta preview, configuring buttons of IKEA (and other remotes) becomes *so much* easier with that.

DrJDizzle, 2018-07-27:
Configuring buttons sound very promising.
Is there a timeline for the new release?

manup, 2018-07-29:
Should be out in the middle of the week.

olemr, 2018-08-21:
Did it work? 
2.05.35 I presume?
Will it be enough just upgrading or will the lights need to be re-added?

DrJDizzle, 2018-08-24:
The switch editor shows up nicely. I can configure the buttons. But only the command for colour temp change doens't show up and cannot be selected for a button.
I will try to readd the lights and switches in the next few days.

closed 2019-01-10 by stale[bot]