dresden-elektronik/deconz-rest-plugin #697: my nodes show 0x0000 and without any name after restore

donnib, 2018-07-23:
[image]

Why is that ?

nero01, 2018-07-23:
I had the exact same, when i connected my new remotes, but the devices were detected correctly after waiting for about an hour.

donnib, 2018-07-24:
@nero01 unfortunately that's not happening for me.

@manup My guess is that since the nodes knew it's gateway they connected to the gateway before i had a chance to restore the backup so now after backup is restored the system is confused ? Am i right in this assumption ? If so how can i solve it ? It would be nice if you had some time to answer this since i don't know what to do now to get my system back. I was hoping the backup/restore would work perfectly.

donnib, 2018-07-24:
@manup ok but i see is the my groups does not include the light they did before the backup/restore. I see all nodes are up but somehow the relation has been messed up. What could be the cause for that ?

manup, 2018-07-24:
Do you mean the group membership of the lights? It is kept in the backup, so when you go in the group > 'manage lights' editor of the Phoscon app you should see the same lights as before the backup?

manup, 2018-07-24:
And do you see the groups in the Phoscon app after restore?
If yes it might take a "while" until the group membership will be restored in the lights. The gateway periodically checks lights that the configured group membership is intact and will correct if neccessary.

It is absolutely possible that you see lights before the restore because the ZigBee network is still running (configuration is active on the shield).

>I guess after step 4 deconz already has the nodes in the database. What happens when i do the restore ? Does the database get wiped ?

The database (zll.db) file will be replaced by the one in the backup file (which is just a tar.gz file).

donnib, 2018-07-24:
@manup i did the restore a day ago. I presume it doesn't take that long ? I see the groups but some of them are empty i mean no lights associated with them.

manup, 2018-07-24:
>@manup i did the restore a day ago. I presume it doesn't take that long ?

No it shouldn't depending on the network size 30&ndash;60 minutes.

>I see the groups but some of them are empty i mean no lights associated with them.

That should not happen either, in the app correct group memberships should be visible instantly after restore.

Which versions of deCONZ are you using before and after restore?

Do you still have the old sd-card? Might be worth a try to copy the files directly.

donnib, 2018-07-24:
I think it was 2.05.31 but based on the Jessie image. The new one is stretch image.

Yes i do have the SD card (i kept it on purpose). So what do you recommend me doing now ? Should i boot the system again with the old SD card then do a backup again then try restore again ? Should i do a reset of the gateway first or ?

manup, 2018-07-24:
>Should i boot the system again with the old SD card then do a backup again then try restore again ? Should i do a reset of the gateway first or ?

2.05.31/32 should work fine, I'll schedules some tests to check the backup feature.

Short guide for manual backup:

* Boot the old sd-card
* Close deCONZ
* Copy  the directory `/home/pi/.local/share/data/dresden-elektronik/deCONZ` to a USB flash drive (or to your computer via WinSCP.

...

* Then boot the new sd-card
* Close deCONZ
* Copy/replace the files with the above ones
* Restart deCONZ

donnib, 2018-07-24:
@manup thx will try

donnib, 2018-07-24:
@manup that did the trick. I noticed not all files are in the dat file. Not everything is backed up ? I also noticed the placement of the nodes in the graphic was not backed up using the backup but doing manually they were retained ?

manup, 2018-11-23:
Backup will be refactored for the next stable release, we identified quite some issues.
Clsoing this for now

closed 2018-11-23 by manup