dresden-elektronik/deconz-rest-plugin #699: Setting schedule time with Phoscon app

ronaldvans, 2018-07-26:
I am trying to create a timed schedule using the Phoscon app. Using the latest Chrome on Windows I can't see the controls in the time box that allow me to set the time. If I click somewhere in the box on the right hand side I can increase/decrease hours and minutes.

Ronoon, 2018-07-26:
Better use the older lightlink app for programming schedules.
So instead of using 
192.168.1.1/pwa/login.html          (phoscon app)
use
192.168.1.1/login.html                  (lightlink app)
there are to many issues this time with the phoscon app e.g. programming, schedules etc.

manup, 2018-07-26:
No please use the Phoscon schedules instead of old webapp since it's the only maintained one.

After clicking in the time field you can modify the time by:
* mouse wheel
* up/down arrows
* directly type the time

ronaldvans, 2018-07-26:
@manup, I tried what you suggested which sort of works. It is not very intuitive and rather hit-and-miss. The old webapp functionality is maybe more basic but it works like a charm!

manup, 2018-07-27:
Agree the time select on desktop has room for improvement (on tablets it should work better there we can use the fancy mobile time widgets).

I'll put it on the list to improve desktop time selection.

closed 2018-07-27 by ronaldvans

xrmthomas, 2019-04-28:
> only Problem is then that you can only turn off the Lights
> how About to turn them scheduled on?
> 
> [image]

This. I need to schedule my lights to turn on. Preferably based upon a dynamic value taken from the virtual daylight sensor.. After that we can worry about scheduling to turn the lights off again.

DAD405, 2019-12-08:
Still Not able to time a Switch on of Lightshow or scenes

mgrn0, 2020-05-23:
Is it really still impossible to switch on lights at a certain time???

AndRe5575, 2020-06-01:
Really poor that such a basic function is still not possible.

Ronoon, 2020-06-01:
Basically just use the Phoscon as a gateway because it has really useful features / integratet components. They are normally very good supportet by @manup or the other contributors to this projekt. 
But the interface sucks !! 
For myself i use Hassio to do the missing programming and timer options.
So look for yourself which option you can use to do your programming or go back to the industrial ones like Phillips HUE if these one is to much of a bother to you.

letszb, 2021-03-08:
> 
> 
> Still Not able to time a Switch on of Lightshow or scenes

You only need to create a scene with "on" for the switch (plug, outlet). You see a list of scenes and the "turn off group" function. If you don't have scenes you only see one option for turon off.