dresden-elektronik/deconz-rest-plugin #701: Support of esp8266 devices

taleteller, 2018-07-27:
Is there any way for deconz to control esp8266 devices, as diyHue does? Like source for a custom firmware to work with. I got a few of those modules and wont like to wrap deconz into another bridge just for getting them to work.

Kane610, 2018-07-28:
Doesn't sound like the purpose of deconz

mark-tech, 2018-07-29:
You don't need another bridge, deconz + diyhue can work together on same device, just follow the information from wiki

taleteller, 2018-07-30:
@mark-tech packing diyhue before deconz is precisely the kind of chaining bridges i want to avoid. deconz has a huge api and plugin support, therefore the question is quite valid if someone has implemented esp8266 bulbs that way.

manup, 2019-01-17:
Closing this for now, for esp8266 devices there are sure better solutions than deCONZ.

closed 2019-01-17 by manup