dresden-elektronik/deconz-rest-plugin #704: Activate scene on motion

oscar-b, 2018-07-31:
Is it possible to configure a scene to activate on motion, using the Phoscon app? I would like to use different colors for the lights during night time and day time for instance.

manup, 2018-07-31:
Currently this is not possible (only via REST API) but already on the near future todo list.

oscar-b, 2018-07-31:
@manup Thanks, will do it via the API for now, thanks! :)

mariusmotea, 2018-07-31:
Depending on the devices that you are using, if are Xiaomi motion sensor of tradfri motion sensor you can put diyHue on top and configure the sensors from hue application were you select the scenes and time frame. Then once the rules from hue app are in place you can even stop diyHue webserver and keep it like a background process, it will just monitor the deconz websocket and apply the scene once motion will be detected. After there will be no motion in selected interval the lights will be dimmed and then in 30 seconds will be turned off. This Philips approach is way much better than Tradfri one, were you can remain in total dark.

oscar-b, 2018-07-31:
Duplicate of #429, sorry didn’t find it before when searching.

closed 2018-07-31 by oscar-b