dresden-elektronik/deconz-rest-plugin #705: Temperature Sensor Developer Manual

ezgicakirofficial, 2018-07-31:
Hello,
I need to find cluster and attributes of Xiaomi Aqara temperature sensor. Is there any developer manual of this sensor?
I found Temperature Measurement cluster (identifies : 0x0402). Is this sensor included in that cluster or are there any other clustures?

manup, 2018-07-31:
For the Xiaomi devices there is no ZigBee specific developer manual, they are quite creative devices only partly standard conform. We have mostly reverse engineered the commands. You may have a look in the REST API sources on how the ZCL attribute reports are handled.

Once the sensor has joined the network it should send ZCL attribute reports to the coordinator when the temperature changes, these are as described in the ZCL specification (Temperature cluster 0x0402). Beside that it also sends some proprietary reports once per hour, with details like battery status.

ezgicakirofficial, 2018-08-01:
However when it has joined the network, it sends commands which belong to basic cluster not temperature cluster. 
Therefore I couldnt get the value of the temperature from this command. Is there any way to get the temperature value from these commands? Can we sen any command to get it or does it send automatically? If it send automatically, I think I couldnt see it.

ezgicakirofficial, 2018-08-01:
Now ı was able to get command from temperature cluster when sensor has joined the network. it sends: 

<Buffer 17 01 00 2c 00 25 00 2a 02 00 00 01 03 _1a 67 a3 01 00 8d 15 00_ 01 04 01 **02 04** 08 _00 18 d7 0a 00 00 29 e9_ 0b 00 af ff a4 00 01 02 f9 28 f8>

As far as ı can parse it, 
-IEEE address is : 1a 67 a3 01 00 8d 15 00
-cluster id : 02 04 (which is temperature measurment cluster)
-asdu : 00 18 d7 0a 00 00 29 e9 

But where can I find to parse the asdu part and get maybe temperature value ?

manup, 2018-08-01:
Almost there, the ASDU part describes a ZCL frame, here it's the ZCL attribute report for the temperature cluster. Note: the sensor will send this periodically on it's own, Xiaomi devices can't be polled.

For the ZCL frame structure in the ASDU have a look into ZCL cluster library specification on these topics:

* General ZCL Frame Format
* Report Attributes Command
* Temperature Measurement

ezgicakirofficial, 2018-08-01:
Thank you. Now I am able to understand the command

closed 2018-12-07 by stale[bot]