dresden-elektronik/deconz-rest-plugin #707: Keen Vents Missing Temperature and Pressure Sensors

brantw, 2018-08-02:
My Keen Vents are just showing up in the Phoscon App as Lights, no temperature or pressure sensors are registered.  The model of my vents is `SV01-410-MP-1.0` and after reading #592 I'm assuming the Keen Vent whitelist needs to be further genericised to support my model of vents and created PR #706.

If you prefer to add a secondary whitelist for `SV01-41` vent models please let me know and I can update my PR.

apmon, 2018-10-19:
Indeed, I just paired another Keen vent (SV01-610-MP-1.4) and that did correctly show up in the API with pressure and temperature.

Btw, those numbers in the version string refer to sizes of the vent. I.e. 610 is a 6x10 inch vent whereas 612 is a 6*12 inch vent

apmon, 2018-11-14:
Attached is a patch to fix the string in the de_we_plugin.cpp to be inline with bindings.cpp. With that fix, I was able to pair my 6x12 vents

[keen_fix.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2582493/keen_fix.txt)

manup, 2018-11-14:
Thanks, have added the patch in the repo.

snjoetw, 2018-11-21:
@manup Is this in the release already? I just tried adding my keen home vents (model # SV01-410-*) and I don't see temp and pressure sensor

snjoetw, 2018-11-21:
Just checked the release page and doesn't look like there's any recent release. Is there a plan to release this soon? Thanks!

manup, 2018-11-21:
Yes the patch is included in latest release 2.05.48

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.cpp#L159

Hanging behind with release notes they will follow in weekend.

snjoetw, 2019-01-21:
Thanks @manup I pulled latest docket image and finally I’m able to see temp and pressure sensors show up. 

But not sure if it’s just mine, have you notice pressure value always stay the same? Same goes for battery level, it’s always 25

fwiw, I’m only using phoscon and these sensors only show up in home assistant but not in phoscon

ksheumaker, 2019-01-30:
I just installed some keen vents, they appear as lights and show up as covers in home assistant but no temperature or pressure sensors.  They appear to have a different model # than what's being detected in deconz and discussed in this issue.  I've attached a screenshot of what phoscon shows.  Can deconz be patched to support SV02-* as well as SV01-*.  Let me know if you need anything else.

[image]

closed 2019-05-31 by brantw