dresden-elektronik/deconz-rest-plugin #710: bulb becomes unavailable but can be controlled through its group

FezVrasta, 2018-08-05:
I have a bulb (actually, a light socket, since I tried to switch them and the same thing happened with the other bulb on the same socket), that sometimes becomes unavailable.

The weird thing is that I can still control it together the other ones in its group if I run some command on the DeCONZ group rather than on the singular bulb.

I guess it's a bug?

olemr, 2018-08-14:
I also see this With some Ikea Tradfri lights.
I'm running version 2.05.32 (beta) headless under Ubuntu.
Seems to work OK for a few days, then need to restart deconz to get it working again.
Group action works OK for on/off and xy color setting, but specifying ct mode after xy does not work for group action.

zilluss, 2018-10-01:
I have the same issue with my Osram Surface Lights and Plugs. I suspect the devices are not added to the "session", because the devices are listed in the GUI and the zll.db, but not the default.session. The Node Info is also not showing the User Descriptor but you can see the MAC Address and the connection indicator.

closed 2019-02-06 by stale[bot]