dresden-elektronik/deconz-rest-plugin #711: View Zigbee mesh with Hass.io on RPi with Raspbee Premium and Deconz add-on

jantje88, 2018-08-06:
I am wondering if it is possible to view the zigbee mesh when using Hass.io on a Raspberry Pi with a Raspbee Premium and the Deconz add-on? Any change to to using using e.g. xming?

jantje88, 2018-09-12:
Any change this is possible @ebaauw ?

ebaauw, 2018-09-12:
Why do you ask me?  I don't know anything about Hass.io nor the deconz add-on for it, but I guess it uses the deCONZ REST API.  There is an undocumented (and unmaintained?) endpoint in the REST API to deliver the neighbour information of a light, but last time I tried that, it caused deCONZ to crash.  See https://github.com/dresden-elektronik/deconz-rest-plugin/blob/9989b3efce58551710af89893060fe1b214d622c/rest_lights.cpp#L1629

jantje88, 2018-09-13:
Saw your posts in many other issues and was hoping you also know something about this question :-)
Thanks for your reply. Maybe this helps others to jump into this. I suppose that I am not the only one with Hass.io and a raspbee and wish to see the zigee mesh.

closed 2019-01-18 by stale[bot]