dresden-elektronik/deconz-rest-plugin #712: OTAU for Ubisys

ebaauw, 2018-08-07:
I found Ubisys released a newer version, 2018Q1, of their firmware (see http://ubisys.de/smarthome/download-firmware.html).  I downloaded it to the ~/otau directory, and it seems deCONZ v2.05.32 with RaspBee firmware 261F0500 picked it up alright.  Initially the upgrade aborted, but it seems it has been applied overnight anyways:
[image]

Checking the OTAU cluster, it seems like the firmware was installed to the dimmer:
[image]

However, the _Basic_ clusters still report the old version (01020105, if memory serves):
[image]

Do I need to do anything to activate the newly downloaded firmware?  I power cycled the dimmer to no avail.  Also, the D1's technical reference gives no clue.

ebaauw, 2018-09-07:
According to the ZCL spec, _Data Code_ is the manufacturing date, not the date of the firmware.

closed 2018-09-07 by ebaauw

ebaauw, 2019-02-20:
The 2019Q1 ubisys firmware is out: http://share.ubisys.de/ubisys-zigbee-firmware-2019-q1.zip.   They're re-doing their website, so it's not yet listed there.  deCONZ seemed happy to install it, in 31:18 minutes.

[image]

ubisys support tell me this version should fix the weird values of _Current Summation Delivered_, as they'll be reporting consumption through their gateway.  At least the dimmer seems to be reporting a reasonable value.

I still cannot make head nor tail from the values reported for _Instantaneous Demand_, _RMS Current_ and _RMS Power_.

manup, 2019-02-20:
Cool hopefully some other routing related issues are also addressed, I'll forwarded it to the team to check.

>I still cannot make head nor tail from the values reported for _Instantaneous Demand_, _RMS Current_ and _RMS Power_.

Do you mean because of the data types?

ebaauw, 2019-02-20:
> Do you mean because of the data types?

No, because of the values.  I'm no electrical engineer, but I would expect _RMS Voltage_ (V) x _RMS Current_ (A) to equal _Active Power_ (W) (modulo rounding/precision errors).  It doesn't.  _RMS Voltage_ is 234.  _RMS Current_ shows 53 when the dimmer and connected lights are off, and 26 when they're on at 100%.  _Active Power_ is 12 when off, and 5 when on.  I've got three 80W 230V halogen lights connected.

manup, 2019-02-20:
Understand, hmm maybe these values need one of those Zigbee transformations similar to lightlevel/lux? Need to check the ZCL spec.

ebaauw, 2019-02-20:
Ubisys support mentioned that the _Current Summation Delivered_ in the previous firmware version contained some internal sensor readings.  I think the other attributes might as well.

The ZCL spec (at least the version I have) is rather brief on this subject.  There’s multiplier and divisor attributes for each, but the D1 doesn’t support those.  There some more of these in the 0x0400 range missing from general.xml, but I don’t think it supports these either.

strandman, 2019-09-19:
Hi, new here!
For Philips Hue Bridge users:
Happy to say that latest ubisys Firmware Version 1.14 (Q2 2019) fix that Philips Hue Bridge recognize the power switch S1 (-R) and S2 (-R) as an "On/Off plug". 
http://www.ubisys.de/en/support/firmware/

OTAU took me only about 6 minutes with ConBee II.

strandman, 2021-04-27:
Hello again!
Ubisys has released new firmwares based on Zigbee Stack 2.0.13
http://www.ubisys.de/en/support/firmware/

But I'm unable to upgrade. The STD OTAU Plugin only stucks at "Queued" and nothing happens.
The change log for S2 states that: 
>  Migration is a one-way process and devices will have a different OTA image type after the upgrade. The new image type is 0x7b23

Could that cause problems? Just guessing? Is there someone else here that is able to upgrade to latest firmware?

Thanks!

SwoopX, 2021-04-27:
The Change log from the J1:

- Migration to new factory block format (FB1)
- Migration to new Persistent Storage 2.0 format and increase of NVM space from 2KB to 20KB
- Migration to ubisys Compact Firmware Upgrade 2.0: CFWU 2.0 is an encrypted, signed and compressed OTA file format. This improves security, cuts update time per device in half with positive effects on battery life for battery powered devices and reduced network bandwidth utilization, and increases available space on the device for future upgrades and new features. We are making all ubisys devices ready for the next major Zigbee release with lots of new features – stay tuned
- Migration is a one-way process and devices will have a different OTA image type after the upgrade. The new image type is 0x7b24
- Added human readable version string
- Added product URL
- Added product code (EAN)

Not too sure if this is good or bad...

SwoopX, 2021-04-29:
So, seems like it worked, but took litereally ages...

[image]

[image]