dresden-elektronik/deconz-rest-plugin #713: Potential battery report issue for IKEA TRADFRI remote control

g-chevalier, 2018-08-07:
Hi,

I started using deCONZ REST API since a couple of weeks, using a Raspbee module.
Just for information, I’m developing a Python plugin to interface with Domoticz, and I start to have some good results.

I noticed with an IKEA TRADFRI remote control ("modelid": "TRADFRI remote control") that the battery level reported is wrong.

With a brand new device, I get in a config report that says:  {"config":{… "battery":47 ….}
I permuted the battery with another device (a Xiaomi pushbutton, "lumi.sensor_switch.aq2" to be exact), and I still have 47 for the IKEA button, and on the Xiaomi button, the same supposed bad battery reports 100.

Any idea?

ebaauw, 2018-08-07:
Don't use Xiaomi devices to compare battery charges.  Best try a new battery instead.  Xiaomi devices don't report battery percentage, only battery voltage.  deCONZ tries to translate this into a percentage, but it's not very reliable (typically it remains at 100% for a long time, then drops quickly when the battery is almost empty).  Also, it might take over an hour for a Xiaomi device to report the battery voltage, so deCONZ might show 100% as a default value for quite some time.

Did you update the firmware on the remote?  That would deplete a fresh battery this much.

What's the firmware version on your Trådfri remote?  In the versions I've seen, IKEA reports the battery percentage in full percents, rather than in 0.5 percents (as per the ZigBee standard).  Maybe they fixed that in a newer firmware?

g-chevalier, 2018-08-08:
In fact, I didn’t wait enough after the battery exchange and reacted too quickly upon the very first notification reports from deCONZ.

After a little while, I finally got a 87 level for the battery: {"config":{"alert":"none","battery":87,"group":"11273","on":true,"reachable":true},"e":"changed","id":"16","r":"sensors","t":"event"}. I don't have a brand-new battery right now, so I can't tell if this value vs 100 % is OK or not. However, it’s far less weird than the former 47 % and seems to be OK.
The fact that Xiaomi button report 100 % for the same battery is another story for which you gave an explanation.

What is important for me is to have a least an alert when the battery is near depletion, especially for sensors like magnet Door / Windows or Motion sensor on which we rely on for alerting more than temperatures or alike.

So this issue can be considered as resolved, and I apologize for having created it too fast!
I know that battery level monitoring is not an exact science, and here we have one more lesson…

Just for the information records:
- My remote is at FW version 1.2.214, and I did not voluntarily update it (I don't know if it can happen automatically).
- Measured battery voltage reporting 87 % on IKEA remote is 3.04 V
- Measured battery voltage reporting 100 % on Xiaomi button is 3.03 V

closed 2018-08-13 by g-chevalier