dresden-elektronik/deconz-rest-plugin #714: Problems adding innr / xiaomi sensors

ssahlender, 2018-08-07:
Hi Together,

to sum up my problem, I don't have success to pair any device with deconz.

First some infos about my setup:

Pi 3 Model B
RaspBee ( quite old, bought it 2014 I think )
Sensors ( bought according to Supported Devices List )
- Innr SP 120
- Xiaomi Aqara weather sensor ( squared one with round corners )
- Xiaomi Smart Plug
deconz:
Version 2.05.32 / 11.7.2018
Firmware 261F0500

At least the Innr SP 120 should work ... the other two from xiaomi I ordered from china ... so maybe I ordered the wrong ones.

To pair I login to the phoscon web app ... devices ... sensors ... add new sensor ... other ... and then I see the screen where it says: the search for new sensors ...

- For the Xiaomi devices I pressed the buttons for 4 seconds and it blinks blue but nothing happens
- For the Innr device I pressed the button for 5 seconds ... some result nothing happens

Finally I get ... Failed to connect

Question 1:
Should there something happen in the webinterface ... or can the sensors only be paired with the REST API?
Question 2:
The xiaomi sensors are not so easy to get in germany, and you never know that you buy the right one ... Does it make sense to put links in the Wiki, or at least article numbers?
Question 3:
Any hints for debugging?
Found this one but it is a little bit too verbose 

Thanks in advance for your feedback.

Regards

Jagohu, 2018-08-07:
Hi,
I have the same issue. For me, it only worked if I completely restarted by RPI with a shutdown, disconnect power cable for 30 sec, reboot. Then straight after reboot it did recognize the Xiaomi sensors, but only for the next couple of minutes. It seemed to work fine with the previous version (28) for me...
For me some sensors also came with a very low battery - that's the other reason why they don't connect. Check the battery, if it's below 3.05 V, consider it low. Unfortunately they fail to report it correctly :(
Good luck!

ssahlender, 2018-08-08:
@Jagohu Thanks a lot for hint ... I replaced the batteries ... and it's working. Thanks a lot.

And about the Innr Plug / Xiaomi Plug ... You have to add it as light not as sensor.

kasteleman, 2018-08-16:
The INNR Plug SP120 can be added as a switch. That is, i use it that way on another platform (Homey). It then supports metering. For now i use polling to get the metering values, because i can't set it with HA profile (https://github.com/kasteleman/com.innr/blob/master/drivers/smart_plug_sp120/device.js).

jo-me, 2018-09-03:
How did you add it so that the energy meter sensors of the innr plug showed up? 
I paired it twice and only the "light" came up afterwards..

closed 2019-01-08 by stale[bot]