dresden-elektronik/deconz-rest-plugin #715: deCONZ runs always on port 80

saxn-paule, 2018-08-08:
Hi,

could you help me to run the deCONZ api on another port then 80?

I don't have a monitor attached so Raspian runs in headless mode.

I already changed the 
/etc/systemd/system/deconz-gui.service to 
`ExecStart=/usr/bin/deCONZ --http-port=8080`

and /usr/bin/deCONZ-autostart.sh to
```
if [ $hassetcap -eq 0 ] && [ "$OS_ID" == "raspbian" ]; then
                ## can use privileged ports?
                setcap -v -q cap_net_bind_service,cap_sys_time=+ep /usr/bin/deCONZ
                if [ $? -eq 0 ]; then
                        OPTIONS+=" --http-port=8080"
                else
                        OPTIONS+=" --http-port=8090"
                fi
        else
                OPTIONS+=" --http-port=8080"
        fi
```

After restarting the Pi top -c still shows
  383 pi        20   0  120148  35144  26828 S   1,6  3,7   0:05.11 /usr/bin/deCONZ -platform minimal --http-**port=80**
  384 root      20   0    5984   3624   2884 S   0,7  0,4   0:00.48 /bin/bash /usr/bin/deCONZ-WIFI.sh
  385 root      20   0    5572   3056   2732 S   0,3  0,3   0:00.41 /bin/bash /usr/bin/deCONZ-update2.sh

manup, 2018-08-08:
To overwrite the headless mode the `deconz.service` file instead of `deconz-gui.service` must be overwritten.

saxn-paule, 2018-08-08:
That did the trick. Thank you.

closed 2018-12-10 by saxn-paule