dresden-elektronik/deconz-rest-plugin #716: Binary links against Qt 5.9

lbschenkel, 2018-08-08:
I am using deCONZ on a Debian (stable = stretch) system and upgraded deCONZ to 2.05.34 (amd64). It now fails to start because the binaries are linked with Qt 5.9, and Debian stretch provides Qt 5.7: https://packages.debian.org/stretch/libqt5core5a. I didn't test the ARM package to check if it broke in Raspbian as well.

Was this deliberate or an oversight?

manup, 2018-08-08:
Hmm damn, yes the latest Ubuntu release was build on 18 LTS, before it was 16 LTS but that machine broke. I'll see to recover the machine and rebuild the package there.

The Raspbian build server hasn't changed, so it should work as before.

lbschenkel, 2018-08-08:
It's fine for me, I didn't know you were using Ubuntu as a base (I presumed it was Debian). I could change my LXC container to use Ubuntu.  But I see benefits of building on Debian stable on AMD64 to be consistent with Raspbian/ARM.

manup, 2018-08-08:
Yeah we use Raspbian and Ubuntu since these are the most common distributions for x86 and ARM. I've updated the package: build with Ubuntu 16 LTS, so everything should be back to normal :)

closed 2018-12-15 by stale[bot]