dresden-elektronik/deconz-rest-plugin #717: Q: Fan control from deCONZ?

ebaauw, 2018-08-08:
Does any-one have any experience with fan control from deCONZ?  I'm looking to smartify my ceiling fans, connecting them to my ZigBee network.

I have regular AC motor fans.  The fan is connected to mains through a single L and N wire pair coming out of the ceiling, powering the motor as well as the lights.  The fan itself has separate L wires for the motor vs the lights.  The fan has two pull cords: one controls the light (on, off), the other the motor, cycling through high, medium, low, and off.  I don't use the light cord: it's always on, as I use smart light bulbs.

If needed, I can pull an additional L wire between the wall switch and the ceiling, but I'd rather not.  I would prefer a solution that would fit in the fan's canopy, like this one: https://www.homedepot.com/p/Hampton-Bay-Universal-Wink-Enabled-White-Ceiling-Fan-Premier-Remote-Control-99432/206591100.  I don't care about the separate RF remote, but the receiver/control unit seems just what the doctor ordered.  However, this control unit only works for 110V fans, and I haven't been able to find a 240V equivalent.  Also, as I already installed smart bulbs in the fan, I don't really need light control. 

My current idea is to try and fit a Xiaomi in-wall switch into the fan's canopy, in between the mains power and motor wires.  While this won't provide speed control, it should enable me to turn the fan on and off from deCONZ (leaving the fan's motor at a pre-set speed).  If the Xiaomi in-wall switch won't fit physically, a ubisys S1 or S2 will, ... at three to four times the price.

I would be nice to be able to control the fan speed from deCONZ as well.  As I understand it, one cannot use a regular light dimmer for this, as that reduces the voltage to the lamp.  Fan control units reduce the amperage instead, keeping the full voltage.

FWillems, 2018-08-08:
Perhaps you can use one of the Sonoff devices like this one.
http://sonoff.itead.cc/en/products/sonoff/sonoff-dual
I don't know how to switch through the different fan speed levels but might be the second channel of the Sonoff dual can solve this.
By flashing the device with the tasmota firmware it should also be possible to emulate hue, but I don't know if deCONZ can recognize that. On the wiki of tasmota firmware it is stated "Wemo and Hue emulation for Amazon Echo (Alexa) support". 
https://github.com/arendst/Sonoff-Tasmota

ebaauw, 2018-08-16:
> Does any-one have any experience with fan control from deCONZ?

Apparently not.  Closing this issue.  For those interested: see my endeavours in https://github.com/ebaauw/ceiling-fan.

closed 2018-08-16 by ebaauw