dresden-elektronik/deconz-rest-plugin #722: Bitron AV2010/34 aka. Climax WSS-4E-ZBS Wall Switch

rwsaxony, 2018-08-10:
Hello everybody,

I have purchased from the Telekom 2 Bitron AV2010 / 34 wall switches, which are in reality Climax WSS-4E-ZBS wall switches.

The device is recognized by the deCONZ and is integrated in the network.
The switch is not recognized by the phoscon.

I am still quite new in deCONZ / Phoscon and have no idea what is needed to get the switch up and running.

Attached the basic things from deCONZ and an xml-file 

Best regards,

Rene

Link to detailed product information:
https://www.zigbee.org/zigbee-products-2/#zigbeecertifiedproducts/productdetails3/58565312b09488924641fad3?ref=view_30_search%3DWSS-4E-ZBS%26view_30_page%3D1

[image]
[image]
[image]

manup, 2018-08-10:
Just a wild guess but it looks like the switch might be able to send group commands in order to recall scenes.
I have one of these in the office, but never testet it, will check next week. Usually the Climax devices are quite easy to integrate.

rwsaxony, 2018-08-10:
Thanks for the support!

rwsaxony, 2018-08-17:
Hello together,

are there any newsworthy findings on the problem?

Kindest regards!

manup, 2018-09-06:
Sorry no updates yet but it is on the near todo list

rwsaxony, 2018-12-07:
Hello again!

just wanted to ask quickly if there are any news regarding this issue?

Kindest regards!

closed 2019-05-23 by stale[bot]