dresden-elektronik/deconz-rest-plugin #729: Keep dropping connection to my Osram lights

MrEaoP, 2018-08-14:
I have a mixed set up of Ikea lamps, sensors & remotes, a couple of Hues and an Osram Garden Pole.

Server :  Raspberry Pi 2 running Raspbian Stretch (headless)
ConBee USB dongle (plugged straight into the USB port)
Version : 2.05.34 / 8/8/2018
Firmware : 261F0500

Vendor : OSRAM
Model: Gardenpole RGBW-Lightify
Version : V1.04.93

Everything was rock solid, but now the Osram Garden lights get dropped (and are turned off) after a couple of minutes. Disconnecting them and waiting a while and then reconnecting them gets them back online for a couple of minutes again.

Checking in on other issues here, it would seem I'm not alone with this type of issues, and that it haw something to do with how the zigbee mesh is built (with a mixed brand infrastructure).

My question is if there is a best practice in which order I power upp the devices, in order to build the mesh correct (so that the Osram talks straight to the USB dongle without relaying via a Ikea or Hue lamp...?) ?

As I'm running headless I can't get (or can I?) those fancy network graphs showing how the mesh is built.

Any advice appreciated.

eiaro, 2018-08-17:
You can get the UI by using VNC Viewer and connecting to your RaspBee

MrEaoP, 2018-08-17:
Thanks for taking the time to try to help.

I am however running Raspbian not only headless, but without a GUI/desktop at all ("Raspbian" vs "Raspberry Pi Desktop").
Adding one afterwords wasn't easy for me either as I got a lot of errors in regard to missing packages...
(So I'll give that a rest for the time being or I guess could give the Pi a complete re-install)

My next step will be to reset the Osram lamp via the touchlink, and then re-adding it without any other lights available, forcing it to bond with the USB-stick. After that I will add lamps, hoping the Osram lights won't try to mesh via the lamps.

eiaro, 2018-08-20:
They will all mesh, hence a mesh network.

I upgraded my Raspbee to a RPi 3+ and it seems to be better for some reason.

MrEaoP, 2018-09-10:
So, as I suspected the Osram was the culprit - got a new one from the store a week ago or so, and it has worked like a charm since.