dresden-elektronik/deconz-rest-plugin #732: Raspbee returns extraneous indication

SwampNinja, 2018-08-14:
I've noticed that with the version of firmware bundled with the latest deconz release (deconz-2.05.34, FW Version = deCONZ_Rpi_0x26210500.bin.GCF), that enqueing a ZDP APSDE request results in two indications coming back - one with a cluster ID in the response without the top bit set, and another with the top bit set. Both indications have the same ZDP sequence number embedded in them.

While I've been able to work around it by looking for the top-bit set version for ZDP requests vs. not for ZCL requests, it seems like a regression since I have not seen that on the original version of firmware I had on my dongle.

Thoughts?

manup, 2018-08-15:
The deCONZ_Rpi_0x26210500.bin.GCF firmware is an enfant terrible doing some nasty things ;)
please use deCONZ_Rpi_0x261F0500.bin.GCF until the next version will be released.

The reason you see two indications is the loopback handler in the firmware. The first indication is the request send from deCONZ (to the node itself), which can be ignored, the second with the high bit set is the response. Firmware prior and after 0x2621 will just send the second indication.

**Note** The high bit set is the ZDP way to mark a response to a request like request: 0x0031 and response: 0x8031 for Mgmt_Lqi_req/rsp.

SwampNinja, 2018-08-15:
Got it - that being said, it would be better to not loopback send the first indication. It's distinguishable when sending ZDP commands, but that won't hold true for ZCL commands.

closed 2018-12-22 by stale[bot]