dresden-elektronik/deconz-rest-plugin #734: Acquire an API key error

rdrzewinski, 2018-08-16:
When I try to generate api key, I'm getting an error: [{"error":{"address":"/","description":"body contains invalid JSON","type":2}}]
I do everything according to the instructions from here: http://dresden-elektronik.github.io/deconz-rest-doc/getting_started/
My deconz version: 2.05.34

ebaauw, 2018-08-16:
What body do you send?

rdrzewinski, 2018-08-17:
Method: POST, URL: http://my-gateway-ip/api. Body is empty.

ebaauw, 2018-08-17:
You should provide a body, specifying the devicetype, see http://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

rdrzewinski, 2018-08-20:
Thanks for help. All works perfectly.

manup, 2018-08-23:
Glad it worked, I'l close the issue for now.

closed 2018-08-23 by manup

Spassat35, 2018-12-30:
Hallo,

Sorry If I need to recall this Challenge one more times. I get exactly the same error. I read the manual multiple times but still don't now how to deal with. 

What I understood is I need to use the Device type and the entry should look like that:
192.168.178.110:80/api?devicetype=???????
is this correct?

and I'm struggling with the device type. in the manual it is described as "Name of the client application". where can I find that? what should I enter here?

Thanks you very much in advance and any help is highly appreciated.

Kind regards

Marcel

Smanar, 2018-12-30:
Hello, Wich one application/client are you using for request ?

192.168.178.110:80/api?devicetype=??????? is a GET request, you need POST request, take a look on previous given links.

ATM you can use what you want for "devicetype"

manup, 2018-12-30:
The devicetype is the name of your application, it can be freely chosen by you like "my app" or "my test script".

* It needs to be provided as a key/value JSON object `{ "key": "value" }`
* It is a HTTP POST request. The object belongs in the body, not the URL.

http://dresden-elektronik.github.io/deconz-rest-doc/configuration/#aquireapikey

#### Example HTTP body
```
{
    "devicetype": "my application"
}
```

Spassat35, 2018-12-30:
Hallo Manuel,

Thanks for the quick response. I didn’t expect this. 
„Belongs to the body“ was the information I just missed or wasn’t clear for me.
Now I tried it one more times and I got the correct response.

Thanks you very much.

Regards
Marcel

Spassat35, 2018-12-30:
Hi, 
Thanks for the quick response. 

I‘m using postman to send the command. Manuell just gave me the right direction.
The device needs to be entered in the Body and this was what I missed.
I tried it and it works now. 

Thanks a lot. 

Regards

Marcel