dresden-elektronik/deconz-rest-plugin #737: REST API to set multiple light states at once?

FezVrasta, 2018-08-19:
Is there any API endpoint to set multiple light states with a single call?

ebaauw, 2018-08-19:
Recall a scene, http://dresden-elektronik.github.io/deconz-rest-doc/scenes/.

FezVrasta, 2018-08-19:
Thanks, but a scene needs to first be defined, I'd need a way to change of a lot of different lights in a dynamic way.

ebaauw, 2018-08-19:
There's no way to do that in the ZigBee standard.

Philips hacked together something specific for Hue Entertainment, see https://developers.meethue.com/entertainment-blog, but that does require support in the light firmware, so it only works with Philips lights.  Not likely to be supported by deCONZ, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/167.

FezVrasta, 2018-08-19:
I see, I just wonder why projects like DIYHue have problems turning on/off a lot of bulbs all together while Hue and DeCONZ bridges can do that just fine 😓

Maybe it's a problem of DIYHue

closed 2018-08-19 by FezVrasta