dresden-elektronik/deconz-rest-plugin #738: 2 questions about Xiaomi Aqara Magic Cube states

B1G1, 2018-08-19:
Hi all, i don't understand how to get the push event on my Xiaomi Aqara cube. In the phoscon app I can see all others state but not the push one.

And... after reading this [post](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/138#issuecomment-325101635) I can't understand what, for example, the 4001 state means? 4 is the face that goes in the bottom after 90° flip? Or what?

ebaauw, 2018-08-19:
You flip from side 1 on top to side 4 on top.  So if the logo is facing you and the cube sits on the side with the battery door, you flip the left side up.  Flipping back yields 1004 (to 1 from 4).

B1G1, 2018-08-19:
If the logo is facing me, the battery door is in the opposite face of the logo

closed 2018-12-24 by stale[bot]