dresden-elektronik/deconz-rest-plugin #739: Loosing Connection to whole Zigbee Network

scobby, 2018-08-19:
Hello Manup,
currently i  loosing the connection to all lamps and sensors, nearly every second day. Then i try restart, unplug, reconfigure everything, search for light,etc, Then with unknown reason it works again. 
This is really annoying. My Girlfriend is *%&$%&%§$/ at me ;-) and its just before i have to remove those smart lights, etc.

The information from the log is not really helpful for the enduser.
I have many gateway´s, servers and technology´s integrated into my homeautomation, but this software is hard as rock. I have some little request, beside my real problem with the lamps.

1. Please clear up those "issues"
Its not helpful if the stay open forever, so nobody can track current errors down and verify if this problem is solved or not. Self-diagnostic is useless (only if you have time to study all the issues).

2. Please create a seperate configuration file for the service. I have to change always the Ports, device, etc.
This is really annoying. 

3. Please create a documentation about the runtime parameters, I only can figure them out with reading issues? (see number 2)

4. remove those not really used services that spamming the system , or maybe remove them from the service "WANTS"-definition.

5. Add a Log-level for Errors and separate logs items into [INFO] [WARNING] [ERROR] [DEBUG]
Add a option into that configuration file where to log and if logging is on which log-level, default to /var/log/deconz/deconz.log maybe?

this is my log from syslog with --dbg-info=1:

Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:30:788 delay sending request 24 dt 1 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:30:839 delay sending request 24 dt 1 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:30:938 delay sending request 24 dt 1 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:31:039 delay sending request 24 dt 1 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:31:139 delay sending request 24 dt 1 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:31:239 delay sending request 24 dt 1 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:31:339 delay sending request 24 dt 1 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:31:439 delay sending request 24 dt 1 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:31:540 delay sending request 24 dt 2 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:31:639 delay sending request 24 dt 2 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:31:738 delay sending request 24 dt 2 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:31:839 delay sending request 24 dt 2 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:31:939 delay sending request 24 dt 2 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:039 delay sending request 24 dt 2 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:139 delay sending request 24 dt 2 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:239 delay sending request 24 dt 2 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:290 delay sending request 24 dt 2 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:339 delay sending request 24 dt 2 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:339 delay sending request 61 dt 0 ms to 0x000B57FFFE10BA90, cluster 0x0004
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:439 delay sending request 24 dt 2 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:439 delay sending request 61 dt 0 ms to 0x000B57FFFE10BA90, cluster 0x0004
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:539 delay sending request 24 dt 3 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:539 delay sending request 61 dt 1 ms to 0x000B57FFFE10BA90, cluster 0x0004
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:638 delay sending request 24 dt 3 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:638 delay sending request 61 dt 1 ms to 0x000B57FFFE10BA90, cluster 0x0004
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:738 delay sending request 24 dt 3 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:739 delay sending request 61 dt 1 ms to 0x000B57FFFE10BA90, cluster 0x0004
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:838 delay sending request 24 dt 3 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:838 delay sending request 61 dt 1 ms to 0x000B57FFFE10BA90, cluster 0x0004
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:938 delay sending request 24 dt 3 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:32:939 delay sending request 61 dt 1 ms to 0x000B57FFFE10BA90, cluster 0x0004
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:33:039 delay sending request 24 dt 3 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:33:039 delay sending request 61 dt 1 ms to 0x000B57FFFE10BA90, cluster 0x0004
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:33:139 delay sending request 24 dt 3 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:33:139 delay sending request 61 dt 1 ms to 0x000B57FFFE10BA90, cluster 0x0004
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:33:239 delay sending request 24 dt 3 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:33:239 delay sending request 61 dt 1 ms to 0x000B57FFFE10BA90, cluster 0x0004
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:33:339 delay sending request 24 dt 3 ms to 0x00158D0001AEB912, cluster 0x0006
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:33:339 delay sending request 61 dt 1 ms to 0x000B57FFFE10BA90, cluster 0x0004
Aug 19 21:54:33 rasp deCONZ[5440]: 21:54:33:355 0x000B57FFFE10BA90 error APSDE-DATA.confirm: 0xE1 on task

scobby, 2018-08-19:
i am on Version:

Hersteller dresden elektronik
Produkt RaspBee
Version 2.05.34 / 8.8.2018
Firmware 261F0500

manup, 2018-08-20:
Oh my, sorry for the trouble at home ;)

Firmware and version looks fine. Can you provide more details about the setup and other services running on the pi? When you loose connection to all lamps and devices something really nasty seems to be going on.

>Please clear up those "issues"
Its not helpful if the stay open forever, so nobody can track current errors down and verify if this problem is solved or not. Self-diagnostic is useless (only if you have time to study all the issues).

You're right here, I think at least issues which weren't updated in a while can be closed. Currently the issue tracker acts mostly as a forum.

>Please create a seperate configuration file for the service. I have to change always the Ports, device, etc.
This is really annoying.

Recommended and clean way is writing a systemd overlay services to customize things like port settings.

>Please create a documentation about the runtime parameters, I only can figure them out with reading issues? (see number 2)

Documentation undergoes a major overhaul currently, to put it short, everything from deCONZ setup, systemd configuration and API will be documented.

>remove those not really used services that spamming the system , or maybe remove them from the service "WANTS"-definition.

The systemd services were improved over the last releases, where possible we will continue to limit various features to be enabled just for the preconfigured sd-card images.

>Add a Log-level for Errors and separate logs items into [INFO] [WARNING] [ERROR] [DEBUG]
Add a option into that configuration file where to log and if logging is on which log-level, default to /var/log/deconz/deconz.log maybe?

There are many log switches for various purposes and ZigBee layers, they will also be part of the new documentation. Writing logfiles is currently not a goal, but you can use journald to do watch logging.

manup, 2018-12-03:
I need more information:
* Which deCONZ version are you using?
* The log shows a channel change, did you manually do this or was is invoked by deCONZ?

You may also send me a backup file so I can have a deeper look. mpi@dresden-elektronik.de

fredronnv, 2019-03-01:
@manup I'd really like to help debug this, I'm having similar issues. I'm running deCONZ in a docker container (marthoc/deconz), latest version. I recently fetched the latest image and my network is unresponsive. Can't control lights at all. Increased debug options and see this:
```
06:27:13:959 APS-DATA.request id: 7, addrmode: 0x03, addr: 0x000b57fffe97c793, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
06:27:15:055 0x000B57FFFE97C793 error APSDE-DATA.confirm: 0xE1 on task
06:27:15:056 APS-DATA.confirm id: 7, status: 0xE1 
06:27:15:359 APS-DATA.request id: 14, addrmode: 0x03, addr: 0x90fd9ffffe29d86c, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
06:27:16:431 0x90FD9FFFFE29D86C error APSDE-DATA.confirm: 0xE1 on task
06:27:16:431 APS-DATA.confirm id: 14, status: 0xE1 
06:27:16:659 APS-DATA.request id: 20, addrmode: 0x03, addr: 0x90fd9ffffe29242b, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
06:27:17:744 0x90FD9FFFFE29242B error APSDE-DATA.confirm: 0xE1 on task
06:27:17:744 APS-DATA.confirm id: 20, status: 0xE1 
06:27:17:959 APS-DATA.request id: 26, addrmode: 0x03, addr: 0x90fd9ffffe2a1dde, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
06:27:19:056 0x90FD9FFFFE2A1DDE error APSDE-DATA.confirm: 0xE1 on task
06:27:19:056 APS-DATA.confirm id: 26, status: 0xE1 
06:27:19:359 APS-DATA.request id: 33, addrmode: 0x03, addr: 0x000b57fffe4a246e, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
06:27:20:448 0x000B57FFFE4A246E error APSDE-DATA.confirm: 0xE1 on task
06:27:20:449 APS-DATA.confirm id: 33, status: 0xE1 
06:27:20:759 APS-DATA.request id: 39, addrmode: 0x03, addr: 0x000b57fffe4da295, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
```

Not really sure where to go from here. Last time this happened I had to completely re-add all lights and sensors, this is something I really would like to avoid.

manup, 2019-03-01:
You may check the *Zigbee configuration* page introduced in 2.05.59.

If the PANID, Channel or Network key have changed, perhaps after updating, you can revert the configuration to an earlier state.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

manup, 2019-03-01:
Oh just notices the 0xE1 errors

>06:27:20:449 APS-DATA.confirm id: 33, status: 0xE1

These refer to MAC channel access failures, when Zigbee module isn't able to send any data because 2.4 GHz channel is too occupied.

Is there any change in surrounding devices like WiFi routers or Audio/Video streaming?

If this is a ConBee a usb extension cable might help.

fredronnv, 2019-03-01:
Hmm, thanks for a swift reply. I am using a USB extension cable.

I will attempt to revert the configuration, if possible and let you know
what I can find out.

I could actually use an ekahau sidekick and perform a spectrum scan to see
if the channel is crowded, but I don't think it is, there haven't been any
changes, that I'm aware of. I will attempt this as soon as I can. Whats
confusing is that messages from motion sensors reach the controller OK,
which I would think indicate that it's not a crowded channel issue.

I'll do some more digging and as I said, let you know.

Fredrik

fredronnv, 2019-03-02:
It was simple, as you hinted, most likely interference from an AC power
cable that was close to the conbee stick itself. No need to get the
spectrum analyser involved.

Curious as to how it determines channel access failures, as the signals
from the motions sensors were received.

Mystery solved though, thank you!

manup, 2019-03-02:
Cool glad it works, interference can be a beast to track.

>Curious as to how it determines channel access failures, as the signals
from the motions sensors were received.

Zigbee uses CSMA/CA which basically translates to listen-before-talk.

Since every node in the network does this, one note might be in a relative quite place and another just a few meters away in a more noisy place (from the nodes point of view). So in your case the sensors likely don't experience the same channel occupation the gateway, which is closer to the cause, does.

closed 2019-07-07 by stale[bot]