dresden-elektronik/deconz-rest-plugin #740: Support for INNR SP 110 smart plug

SkorpionNL, 2018-08-20:
I saw that the newest INNR smart plug (SP 120) can be paired with Deconz, but unfortunately the older one (SP 110) not at the moment.

Is someone willing and able to add support for the SP 110?

kasteleman, 2018-08-20:
> Haven't come across the SP 110 before, but I expect that would already be
> exposed as a /lights resource. If not (maybe it's a RFD?), could you post
> screenshots of the deCONZ GUI for:
>
>    - The node, including its endpoints and clusters;
>    - For the *Node info* panel;
>    - For the *Basic* cluster in the *Cluster info* panel?

It can be used as a switch with zigbee HA 1.2
Attribute reporting of the on / off is supported.
Implemented it for ZigBee support within Homey

SkorpionNL, 2018-08-20:
Unfortunately I don’t have access to the GUI due to the use of the hassbian image. I can confirm that the SP 110 works on Homey. Is the info that @kasteleman provided sufficient for adding the smart plug to Deconz?

ebaauw, 2018-08-20:
No, we need the info from the screenshots I mentioned earlier:

Node:
- Is it an end device or a router?
- What’s the mac address prefix?
- Which endpoints does it have?
- Which client vs server clusters does each endpoint have?

_Info panel_:
- Is it an end device or a router?
- What’s the mac address prefix?
- What’s the _Manufacturer Id_?

Basic cluster:
- What’s the _Manufacturer Name_?
- What’s the _Model Id_?

kasteleman, 2018-08-20:
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] Node: 354fdd49-f3c7-4b73-ac1c-f5d3f26035cb
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] - Battery: false
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] - Endpoints: 0
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] -- Clusters:
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] --- zapp
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] --- genBasic
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- cid : genBasic
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- sid : attrs
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- zclVersion : 1
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- appVersion : 2
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- stackVersion : 2
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- hwVersion : 1
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- manufacturerName : innr
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- modelId : SP 110
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- dateCode : 2015-08-25
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- powerSource : 1
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] --- genIdentify
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- cid : genIdentify
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- sid : attrs
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- identifyTime : 0
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] --- genGroups
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- cid : genGroups
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- sid : attrs
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- nameSupport : 0
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] --- genScenes
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- cid : genScenes
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- sid : attrs
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- count : 0
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- currentScene : 0
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- currentGroup : 0
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- sceneValid : 0
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- nameSupport : 0
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- lastCfgBy : 0xffffffffffffffff
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] --- genOnOff
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- cid : genOnOff
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- sid : attrs
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- onOff : 1
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] --- seMetering
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- cid : seMetering
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- sid : attrs
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- currentSummDelivered : [ null, 0 ]
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- status : 0
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- unitOfMeasure : 0
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- summaFormatting : 37
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ---- meteringDeviceType : 0
2018-08-20 19:25:01 [log] [ManagerDrivers] [smart_plug] [0] ------------------------------------------

Awaiting my conbee to arrive, so this is the info I get within Homey.

Does that help?

ebaauw, 2018-08-20:
A bit.  I've edited my previous post to keep track of the still missing info.

kasteleman, 2018-08-20:
Ah forgot to mention that it is a router. Also, manufacturer id is the same
as the SP120

ebaauw, 2018-08-20:
> Ah forgot to mention that it is a router

Then it should already be exposed as a `/lights` resource.  Can you post what the resource looks like (just a GET of /lights/xx).

kasteleman, 2018-08-20:
I can’t at this moment, because i’m awaiting my conbee to arrive ;-)

Maybe skorpion can?

SkorpionNL, 2018-08-21:
The smart plug isn't listed when I do a get request of /lights. But that's as expected because I can't add the plug to Deconz right?

ebaauw, 2018-08-21:
You should be able to add it to deCONZ.  It's gonna be hard debugging without the deCONZ GUI to check whether the node is there.  Reset the plug (I guess, hold the button for 10 seconds), open the network from the web app or search for lights on Phoscon.  You probably need to power cycle the plug (unplug and re-insert it).

SkorpionNL, 2018-08-21:
I tried everything yesterday and today, but unfortunately no luck. Plugging in/out, turning off/on, turning on/off 6x,7x times, holding the button for multiple seconds. Tried with phoscon and the web app, even at very close range of the raspbee.

Will await the arrival of @kasteleman’s Conbee. Hopefully he has more luck.

SkorpionNL, 2018-08-22:
The plug itself seems to be the culprit. Adding it to Homey doesn’t work anymore as well. The only way is finding out how to reset the plug I guess.

SkorpionNL, 2018-08-22:
I reset the plug and added it successfully to deconz thanks to @kasteleman. Turned out the plug must be reset with a magnet. Who figures.. Will close this issue.

closed 2018-08-22 by SkorpionNL