dresden-elektronik/deconz-rest-plugin #742: Support for MiLight

saxn-paule, 2018-08-21:
Is it possible to support the MiLight protocol aswell to get rid of the MiLight bridges?

ebaauw, 2018-08-21:
Technical information about these seems very hard to get.  Do they even talk ZigBee?

saxn-paule, 2018-08-21:
Unfortunately not, they use their own proprietary protocol but use the 2,4GHz frequency as well.

But the protocol was already reverse engeneered and there are own implementations/emulations of the gateway available.

https://hackaday.io/project/5888-reverse-engineering-the-milight-on-air-protocol

https://blog.christophermullins.com/2017/03/18/reverse-engineering-the-new-milightlimitlessled-2-4-ghz-protocol/

https://blog.christophermullins.com/2017/02/11/milight-wifi-gateway-emulator-on-an-esp8266/

ebaauw, 2018-08-21:
I don’t think it’s technically possible.  A lot of different protocols use 2.4GHz (because that band can be used world-wide without a broadcasting license), WiFi, Bluetooth, ZigBee, but you need different hardware for each of these, or at least different radio firmware.

closed 2018-12-23 by saxn-paule