dresden-elektronik/deconz-rest-plugin #744: Xiaomi aqara wall switch clusters info

ezgicakirofficial, 2018-08-22:
Hello,
I am trying to get commands from and send commands to xiaomi wall switch. As far as I saw it has two clusters: on/off and multistate output. Is there any other cluster you know?
Additionally, when I worked with other xiaomi zigbee devices, for ex. aqara door sensor, if I want to get initial information about the device or make it join the current opened network, I was pressing its button for 10 secs and I was sending some commands which belongs to basic or multistate output. But I couldnt have such a button of wall switches. Do you know how can I get those inital commands and make it join the network?
Thank you in advance.

manup, 2018-08-23:
Xiaomi devices do have also a basic cluster and will send some of it's attributes like `modelid` on join. Sending ZCL commands to the devices never really worked well, so I wouldn't recommend doing that, however they send their state updates very reliable, so querying the devices shouldn't be necessary.

The wall switches don't have a dedicated setup button but when (all) the normal buttons are pressed for some seconds (10?) the switch should do a reset and join to the network.

ezgicakirofficial, 2018-08-24:
Thank you for your answer. I was helpful. I have another question also. I can get the on/off state of the switches however I want to switch the light on by sending commands like we press the switch's button and make it on. Commands belonging to On/Off cluster does not do that ı think. When I send command set to on, it returns a command that indicates its state is off. Do you know how I can implement such a function ?

manup, 2018-08-24:
The Xaiomi switches are very limited and non-standard in that regard, they only provide the information that a button was pressed (when I recall correctly they don't support released event) via on/off server cluster. Therefore to control lights you need to send the related commands from your code after receiving the button was pressed event. In deCONZ we use the rule system for that.

Note: Other switches like from IKEA and Philips support configuring them so that lights can be controlled directly from the switch.

ezgicakirofficial, 2018-08-24:
> Therefore to control lights you need to send the related commands from your code after receiving the button was pressed event. In deCONZ we use the rule system for that.

I dont understand what you meant here. I want to send related commands, yes. But why after receiving button was pressed event? 
After I pressed button I am able to see some commands that indicate whether the switch on or off. I am able to parse them, yes. But I cannot change the status of switch by sending commands. I tried to send commands belonging to on/off cluster and manipulate the of/off attribute of switch, it didnt work. 
Do you mean some other cluster commands by related commands? If you do, do you know which clusters they are?

ezgicakirofficial, 2018-08-24:
mine is mains powered. As far as I saw on deCONZ it has two different endpoints. I sent commands to both endpoint but it didnt work also. I wish I know which commands ı should sent. :/

ebaauw, 2018-08-24:
There’s at least three different wired models, each of which come in a one-gang and a two-gang variant.  Do you know the _Model indentifier_ as reported in the _Basic_ cluster on the first endpoint?

The tend to have three different types of endpoints, see also 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/356:
- For controlling the swithed output wire (L1, L2).  Basically an _OnOff_ cluster which accepts _On_, _Off_, and _Toggle_ and reports on _OnOff_;
- For power measurement.  An _Analog input_ cluster.  One endpoint reports total consumption, the other current power.  It reports _Present Value_.
- For buttonevents (one for left button, one for right button, and one for both buttons).  A _Multistate input_ cluster and an _OnOff_ cluster.  It reports _Present value_, not sure about these _OnOff_ clusters.

ebaauw, 2018-08-24:
Unfortunately, the link doesn’t help - they all look the same.  It it a router or an end device?  Does it have a connection for a neutral wire (N)?

I think the _Multistate input_’s _Present value_ reports 0 when the button is not pressed, and 1 when it is, but I need to check the code.  Not at home right now.

ezgicakirofficial, 2018-08-24:
Oh, sorry. model identifier is lumi.ctrl_neutral1. Does it help? 
when click the buttons ı cant see any difference on Multistate ınput's present value. 
There are lots of endpoints and I couldnt figure out which one to manipulate. You can find the screenshot of deCONZ gui.
[image]

ebaauw, 2018-08-24:
> Oh, sorry. model identifier is `lumi.ctrl_neutral1`. Does it help?

Yes!

For this model, endpoint 02 is for controlling the L1 output, see https://github.com/dresden-elektronik/deconz-rest-plugin/blob/2bd22c636d8a920021e7672786daf25cdbf6c78e/de_web_plugin.cpp#L1248-L1257

and

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/2bd22c636d8a920021e7672786daf25cdbf6c78e/de_web_plugin.cpp#L1282-L1288

(Sometimes the device ID isn't read properly, hence the two checks.)  The connected light should react on issuing _On_, _Off_, or _Toggle_ commands to this endpoint.  Additionally, the _OnOff_ attribute should reflect the correct state (Xiaomi does attribute reporting to the coordinator out of the box).

Endpoint 03 is for the L2 output of the `lumi.ctrl_neutral2` (they have more or less the same firmware, I assume).  It doesn't do anything for the `lumi.ctrl_neutral1`.

Endpoint 04 should report the button presses - not sure how useful this is, since the button is hard-wired to the L1 output.  This model uses the _OnOff_ cluser, see
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/2bd22c636d8a920021e7672786daf25cdbf6c78e/de_web_plugin.cpp#L2849-L2855

The value of the _OnOff_ attribute mimics the value of endpoint 02, so whenever it changes, the button has been press/released, see
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/2bd22c636d8a920021e7672786daf25cdbf6c78e/de_web_plugin.cpp#L4764-L4775
Endpoint 05 and 06 are for the second button and both buttons on the `lumi.ctrl_neutral2`.

Endpoint 08 should report the total consumption, see https://github.com/dresden-elektronik/deconz-rest-plugin/blob/2bd22c636d8a920021e7672786daf25cdbf6c78e/de_web_plugin.cpp#L3003-L3009
Not sure if this model actually reports anything useful here, though.

ebaauw, 2018-08-27:
Could also be a routing problem.  I recently added two `lumi.ctrl_neutral2` switches to my production network (now 89 nodes), but occasionally they’re not responding.  I suspect the RaspBee firmware (261f0500) has the wrong route to the end-device, causing commands to get lost.  I added an XBee to the network as well, and see XCTU sometimes reporting a different parent for the switch than deCONZ.  Sometimes deCONZ even shows multiple parents, even though I didn’t powercycle any node.  This would also explain why my Hue motion sensors occasionally blink red and sometimes even drop off the network.

I would check:
- Can you control another (real) light from through the serial port?
- Can you control the switch from the deCONZ GUI?  If not, power down all routers, forcing the switch to connect to the RaspBee/ConBee.

ezgicakirofficial, 2018-08-29:
I think they are not the problem that I have. I also tried to connect them with ConBee. It still works on deCONZ but does not work with commands that I am sending explictly. 
As I said before I monitored serial port commands and found these differences between my command and deCONZ: 

- For ex, deCONZ sends this :  c0 03 5b 00 18 00 11 00 fc 02 c3 5b 02 04 01 06 00 01 03 00 01 fc 01 00 00 4e fc c0. According to deCONZ serial protocol documantation,  01 fc 01 part is the payload. Its size is 3. Hovewer in zigbee cluster library specification it says "on" command belonging to on/off cluster does not have payload. That is why I was sending the payload part as "01", which is command id of "on" command. Do you have any idea reason behind this issue?

- deCONZ sends commands by specifying nwk address of device but I am sending by MAC address. I cannot decide if it will cause a problem.

- According to deCONZ serial protocol, there must be "flags" field which is set to 0 in these commands but deCONZ does not put this field in commands at all. I was putting them. 

Considering these differences, is there anything seems to cause a problem? Thanks in advance.

ezgicakirofficial, 2018-08-29:
I fixed the problem. I must have written the asdu respect to command frame formats in the zigbee cluster library specification. Now it works, thank you for your help.

manup, 2018-10-05:
Closing this issue, discussion continues in issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/798

closed 2018-10-05 by manup

bramschats, 2019-08-17:
I added the Xiaomi wall switch to my DeCONZ instance but after connecting the device I can only see my battery sensor within home assistant and I don't have the 0006 on/off cluster. I think that is why the switches aren't working correctly! Is this a software bug?

Smanar, 2019-08-17:
You can't see cluster 0006 in deconz too ?
If yes, try to pair it again, without deleting them.

bramschats, 2019-08-18:
Tried to repair without deleting the device doesn't make a difference, zee picture.

[image]

ebaauw, 2019-08-18:
>  the Xiaomi wall switch 

"The" ???  There's at least a dozen different models Xioami wall switches.  Looks like you have a wireless switch.  What _Model Identifier_ does it report in the _Basic_ cluster?

bramschats, 2019-08-18:
The device list shows:

>Vendor Xiaomi
Product Transmitter 2-gangWXKG02LM Rev.2
Version 20180809

Smanar, 2019-08-19:
It can be something normal, your device can use the cluster 0012 instead of 0006
It don't work on Phoscon ?

Jbb08, 2020-01-04:
Hello @ebaauw I have just installed a new Xiaomi Aqara Wall Switch (wired no neutral) Single Button. I am new to Deconz/Phoscon and my Conbee 2 (ConBee II Version: 2.05.72 Firmware: 264A0700) and I added the switch with great confusion.
First in Phoscon I tried adding it as a Switch, but it would not find it. Then I tried light, and it found it but kept coming up with 8 lights, after restarting everything I now have 2 items under Lights in the app, one shows as a power outlet, and the other as a light on/off 1.
Once I exposed these in Homebridge hue, now in Home App I have 2 accessories, one I named Bathroom Lights, the second is that light on/off 1, Bathroom lights when toggled turns the switch on and off, the other one does nothing... 
I am really confused, can you shed some light as to why this might be happening, when ever I delete the on/off 1 from phoscon it removes the Light (showing as an outlet) so I have to keep both to keep it working.. is there a way to not expose the on/off 1 in hue config? Or have I don't something wrong.

Many thanks for your help in advance.
[image]
[image]
[image]

ebaauw, 2020-01-04:
These are wired switches to smartify dumb lights.  They behave like lights, and are exposed as `/lights` resources.  Very different from wireless switches used to control smart lights, which are exposed as `/sensors` resources.

I think the REST API plugin still incorrectly exposed two `/lights` resources for the `lumi.ctrl_neutral1`, for endpoints 0x02 and 0x03?  homebridge-hue exposes these as a single accessory with two _Lightbulb_ services.  With the latest iOS, the Home app might show that as one or as two tiles, depending on the accessory settings.  For older iOS and on macOS it shows two tiles.

To stop homebridge-hue from exposing the non-functional resource for endpoint 0x03, create a blacklist resourcelink on deCONZ containing that resource.  Make sure to check the `uniqueid` of the resource: it should end with `-03` (for the endpoint).  See https://github.com/ebaauw/homebridge-hue/blob/fd9bc82a6d5df19fd516e02955244c4e24f03170/README.md#configuration.

Jbb08, 2020-01-04:
Thanks @ebaauw I guess that confirms that it is exposing it incorrectly, as an outlet and light weirdly. I can have it as a group of tiles, or as single tiles on my iOS (if I create a group, when its on, and the physical switch is used, the group only turns off the light not the extra resource, causing the group to be out of sync) I appreciate that you linked to how Homebridge-hue could use a black list, but I have no clue how to actually create this on my gateway in deCONZ. Is this done via the deCONZ GUI or terminal. Where can I find my uniqueid for the correct resource to blacklist. Is there a way you can step me through how to do this? I have no idea how to create a resourcelink on deCONZ
Many thanks.

ebaauw, 2020-01-04:
The resource link is created through the API, by doing a POST to `/api/`_apikey_`/resourcelinks` and a body as in the README.  To find the ID of the `/lights` resoucre, do a GET of `/api/`_apikey_`/lights`.

I use `ph` to interact with the API from the command line.  This little utility is included with homebridge-hue.  Issue `ph -h` from the command line for help.

If you're on the host running deCONZ, you would do something like:
```
$ ph createuser
"6A11EC624F"
$ ph get -al /lights | grep '/uniqueid:"00:15:8d:.*-03"'
/326/uniqueid:"00:15:8d:00:02:b5:4b:93-03"
/422/uniqueid:"00:15:8d:00:02:39:14:5e-03"
/424/uniqueid:"00:15:8d:00:02:4b:d5:7e-03"
$ ph post /resourcelinks '{
{
  "name": "homebridge-hue",
  "classid": 1,
  "description": "blacklist",
  "links": [
    "/lights/326"
  ]
}'
1
$
```
Make sure to check the mac address of the `lumi.ctrl_neutral1` in the response to the GET and substitute the ID of the resouce in the body of the POST.  326, 422, and 424 correspond to the second button on my `lumi.ctrl_neutral2` switches.

The api key from `ph createuser` is saved in `~/.ph`, so you needn't remember it yourself.  Also you don't need to remember the ID of the resourcelink, as returned by the POST.

When you restart homebridge with `-D`, you should see mesages like:
```
[1/4/2020, 1:38:57 PM] [Hue] pi2: /resourcelinks/1: 1 blacklist entries
[1/4/2020, 1:38:57 PM] [Hue] pi2: /lights/326: blacklisted
```

Jbb08, 2020-01-04:
You are amazing @ebaauw thank you so much, I understood about 60% of what you wrote hehe
In the end my setup was as follows - Pi 4b running deCONZ, Homebridge hoobs using config ui x. Used terminal to access the api via ph but no luck, 404 errors (See below). 
So I went and read the REST API documentation and used Chrome with Postman plugin to access the REST API, and managed to get an API key, and run all the commands (modified) you gave me above, which was so very helpful. I want to say thank you so much :) It al worked wonderfully.

One last question, as you see below my devices from the lights/ GET are shown without any details oddly, can I add "manufacturer name" "modelid" "swversion" and change the type from "Smart Plug" details to change it from Unknown/Null?

Thanks again 👍 
[image]
[image]
[image]

ebaauw, 2020-01-04:
> Used terminal to access the api via ph but no luck, 404 errors

404 means: not found.  You're running another web server on port 80 instead of deCONZ.  Use `ph -H `_hostname_`:`_port_ or set the `PH_HOST` environment variable to the _hostname_`:`_port_ that the deCONZ API runs on.

> One last question, as you see below my devices from the lights/ GET are shown without any details oddly, can I add "manufacturer name" "modelid" "swversion" and change the type from "Smart Plug" details to change it from Unknown/Null?

Looks like the _Basic_ cluster hasn't been read by deCONZ.  In the GUI, select the _Basic_ cluster of endpoint 0x01 of the switch.  Then in the _Cluster Info_ panel read the attributes.  That should populate `modelid`, `manufacturername` and `swversion`.  I don't know where the `type` value `"Smart Plug"` came from.  It should be populated from the device type on the endpoint.  According to your screenshot, both endpoint 0x02 and 0x03 report `"On/Off light"`.  You might try and read the _Simple Descriptor(s)_ from the left dropdown circle in the node, but I'm not sure if the REST API plugin will update `type` once the resource has been created.  If not, re-pair the device or (not for the faint hearted) patch the database.