dresden-elektronik/deconz-rest-plugin #745: IKEA Trådfri motion sensor and remote control issue in V02_05_35

Maaacus, 2018-08-23:
Hi! 

A few days ago I restored my Raspberry Pi 3 b+ and reinstalled my Home Assistant. Installed the latest Deconz V 02 05 35 and paired my IKEA Trådfri lights without any issue. However I couldn't pair the motion sensor and remote controll. Tried to downgrade the software/firmware, reinstall and so on. Nothing solved it. After that I decided to try install my latest version which worked a couple of weeks ago. Reinstalled V 05 32 and the pairing worked like a sharm again. 

Is this a known issue? Anyone else with the same problem?

Best regards,

balu-, 2018-08-24:
it seems that i have the same problem, 
was migrating deconz to a new pi, lights & philips motion sensors paired without a problem.
I could not get tradfri motion & remote to pair.

manup, 2018-09-01:
You may try 2.05.37, there are some changes to correct invalid and old data about devices in `zll.db` and `session.default` files to mitigate the above problems.

balu-, 2018-09-03:
Thanks, @manup the new version seems to fix my problem 👍

closed 2018-09-04 by manup