dresden-elektronik/deconz-rest-plugin #747: Problem with the rest-plugin to update ikea trådfri

mrlook2, 2018-08-23:
Hi all ! 

Running a virtual ubuntu 18.04 with desktop just to be able to uppdate the firmware in my ikea trådfri bulds. But I.m having som problem. Everythig works fine untill I'm installing the rest API.

`look2@deconz:~$ sudo dpkg -i deconz-2.05.35-qt5.deb
Selecting previously unselected package deconz.
(Reading database ... 175085 files and directories currently installed.)
Preparing to unpack deconz-2.05.35-qt5.deb ...
Unpacking deconz (2.05.35) ...
Setting up deconz (2.05.35) ...
Processing triggers for gnome-menus (3.13.3-11ubuntu1.1) ...
Processing triggers for desktop-file-utils (0.23-1ubuntu3.18.04.1) ...
Processing triggers for mime-support (3.60ubuntu1) ...
look2@deconz:~$ sudo apt install -f
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.`

and all is working fine
The we go for the rest plugin:

`look2@deconz:~$ git clone https://github.com/dresden-elektronik/deconz-rest-plugin.git
Cloning into 'deconz-rest-plugin'...
remote: Counting objects: 5769, done.
remote: Compressing objects: 100% (122/122), done.
remote: Total 5769 (delta 69), reused 179 (delta 41), pack-reused 5556
Receiving objects: 100% (5769/5769), 5.84 MiB | 6.98 MiB/s, done.
Resolving deltas: 100% (3962/3962), done.
look2@deconz:~$ cd deconz-rest-plugin
look2@deconz:~/deconz-rest-plugin$ git checkout -b mybranch V2_05_35
Switched to a new branch 'mybranch'
look2@deconz:~/deconz-rest-plugin$`

No errors while compiling, and then I do the copy: 
look2@deconz:~/deconz-rest-plugin$ sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins

but after that I can't access the program, or the web interface. when trying to look at the service:
`look2@deconz:~/deconz-rest-plugin$ systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; disabled; vendor preset:
   Active: failed (Result: exit-code) since Thu 2018-08-23 14:31:29 CEST; 24min
  Process: 1684 ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 (code
 Main PID: 1684 (code=exited, status=143)

aug 23 14:28:44 deconz systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
aug 23 14:28:44 deconz deCONZ[1684]: libpng warning: iCCP: known incorrect sRGB
aug 23 14:28:45 deconz deCONZ[1684]: This plugin does not support propagateSizeH
aug 23 14:28:45 deconz deCONZ[1684]: This plugin does not support propagateSizeH
aug 23 14:30:53 deconz deCONZ[1684]: This plugin does not support propagateSizeH
aug 23 14:31:29 deconz systemd[1]: Stopping deCONZ: ZigBee gateway -- REST API..
aug 23 14:31:29 deconz systemd[1]: deconz.service: Main process exited, code=exi
aug 23 14:31:29 deconz systemd[1]: deconz.service: Failed with result 'exit-code
aug 23 14:31:29 deconz systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.

look2@deconz:~/deconz-rest-plugin$ systemctl stop deconz
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to stop 'deconz.service'.
Authenticating as: oscar,,, (look2)
Password:
==== AUTHENTICATION COMPLETE ===
look2@deconz:~/deconz-rest-plugin$ sudo systemctl start deconz
look2@deconz:~/deconz-rest-plugin$ sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-08-23 14:59:02 CEST; 7s ago
  Process: 10957 ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 (code=exited, status=127)
 Main PID: 10957 (code=exited, status=127)

aug 23 14:59:02 deconz systemd[1]: deconz.service: Service hold-off time over, scheduling restart.
aug 23 14:59:02 deconz systemd[1]: deconz.service: Scheduled restart job, restart counter is at 6.
aug 23 14:59:02 deconz systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
aug 23 14:59:02 deconz systemd[1]: deconz.service: Start request repeated too quickly.
aug 23 14:59:02 deconz systemd[1]: deconz.service: Failed with result 'exit-code'.
aug 23 14:59:02 deconz systemd[1]: Failed to start deCONZ: ZigBee gateway -- REST API.

look2@deconz:~/deconz-rest-plugin$ /usr/bin/deCONZ -platform minimal --http-port=80
libpng warning: iCCP: known incorrect sRGB profile
/usr/bin/deCONZ: symbol lookup error: /usr/share/deCONZ/plugins/libde_rest_plugin.so: undefined symbol: sqlite3_open`

What am I doing wrong?

manup, 2018-08-23:
Looks like development libs for sqlite3 aren't installed.

mrlook2, 2018-08-23:
gues not.

look2@deconz:/lib/systemd/system$ sudo apt install libsqlite3-dev
[sudo] password for look2:
Reading package lists... Done
Building dependency tree
Reading state information... Done
libsqlite3-dev is already the newest version (3.22.0-1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.

manup, 2018-08-23:
Another guess, when I remember correctly on Ubuntu you need to install the pkg-config package, qmake uses this to find the sqlite3 libraries.

mrlook2, 2018-08-23:
So I should install pkgconf and the recompile ?

manup, 2018-08-23:
Yes if  `$ pkg-config --libs sqlite3` on the commandline doesn't work it needs to be installed.

mrlook2, 2018-08-24:
$ pkg-config --libs sqlite3
-lsqlite3

is what I get

manup, 2018-08-24:
Hmm strange in that case it means it is already installed as well as libsqlite3-dev which should be enough for qmake to do the proper linking.

By reading the top post I wonder do you really need to compile the plugin on your own? It is already included in the deCONZ package and you only need to compile it if you indent to modify it.

mrlook2, 2018-08-24:
I followed the instructions on the first page. 
I did it once again and then it worked when I recompiled it it worked.

closed 2018-12-29 by stale[bot]