dresden-elektronik/deconz-rest-plugin #750: Lights not being discovered anymore...

SwampNinja, 2018-08-24:
I have a couple of Phillips Hue bulbs and an IKEA Tadrfi bulb as well, and all have been playing nicely with the Raspbee for a while now. Recently however, I can't seem to discover the Phillips bulbs anymore. I managed to discover the IKEA once after a couple of tries, but even then, it was a struggle.

I'm literally 2 feet away from the bulbs.

The dongle is running firmware 261F0500 and I have Deconz version 2.05.34.

Webserve, 2018-08-26:
Try to reset the bulbs with the remote.

https://github.com/mozilla-iot/wiki/wiki/HOWTO:-Factory-reset-a-Hue-bulb

SwampNinja, 2018-08-27:
I was able to reset them using the old Phoscon web app that supports Touchlink (see [this](
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/662#issuecomment-401023834))

closed 2018-08-27 by SwampNinja