dresden-elektronik/deconz-rest-plugin #751: Phillips Hue - Attribute Reporting

SwampNinja, 2018-08-25:
I tried to setup attribute reporting for both the On/Off attribute in the On/Off cluster as well as the CurrentLevel attribute in Level cluster on my Phillips A19 lightbulb, and both returned with an error of "UNREPORTABLE_ATTRIBUTE". This despite the fact that both of these attributes look to be required to be reportable in the ZCL spec. Are they just violating the spec or am I reading the spec wrongly?

ebaauw, 2018-08-25:
The ZLL spec actually specifies there’s no attribute reporting for these.  I suspect Philips had a good say in that.

SwampNinja, 2018-08-26:
Bummer, that’s a real disappointment.

I tried with the IKEA Tradfri bulb as well, and while it doesn’t puke on the request, it returns an invalid response (just a single byte in the ZCL response instead of a full “Configure Reporting Response”).

Even then, I don’t see any frequent indications (I had set the minimum to 2s, max to 4s, minimum change to 1).

Any ideas?

ebaauw, 2018-08-26:
Ikea bulbs do support attribute reporting, but the setting seems to be lost on a power cycle.  I find setting up attribute reporting in the deCONZ GUI doesn’t always give feedback, but when reading the reporting settings afterwards, they seem to be stored anyways.  Also, the REST API plugin creates the binding before setting up attribute reporting, allegedly because some devices require this order.

Typically I set _OnOff_ to min 1, max 300, and _Level_ and colour attributes to min 1, max 300, change 1.  See 
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/2bd22c636d8a920021e7672786daf25cdbf6c78e/bindings.cpp#L733

SwampNinja, 2018-08-27:
I’m actually writing my own ZigBee application stack that interacts with the dongle directly over serial, so a lot of this I have to implement on my own :P

I was able to get it all working like a charm after setting up a binding!

closed 2019-01-01 by stale[bot]