dresden-elektronik/deconz-rest-plugin #752: Connect switch from manufacturer not listed

JAB33699, 2018-08-25:
Hi,
is there a possibility to connect a switch from manufacturer who is not listed in the phoscon app?
Cheers Jürgen

mat1990dj, 2018-08-26:
It would be nice to do a custom mapping and at least have the switches on the rest api (not needed to be on the phoscon webapp)

manup, 2018-08-26:
Is it a specific switch you have in mind? Currently the REST-API supports more switches than shown in the Phoscon App, to pair them you can start sensor search in the Phoscon App.

Switches can't be supported in a general way by the API since we need to add custom code which is specific to the switch/sensor. However we strive to support most of the popular ones.

mat1990dj, 2018-08-26:
Well, I bought two trust (klikaanklikuit) motion sensors, and two remotes (ZYCT-202) and I've indeed paired it with my raspbee using "Other Sensor" search method (as in deconz program they appear the 4 of them linked) but none of them is appearing neither on the REST-API nor the phoscon webapp.

mat1990dj, 2018-09-07:
Is there anything that I can do to help on the development?

manup, 2018-09-10:
>Is there anything that I can do to help on the development?

Most helpful are screenshots of the nodes, `cluster info panel` of basic cluster with the model identifier and `node info panel`.

Further details of received packets are provides with `--dbg-aps=2 --dbg-zcl=2` deCONZ commandline parameters.

jee67, 2018-10-04:
I’m a step further. Both my ZYCT-202 were connected to deCONZ. They are still not visible in the Phoscon App, but my domotica system sees them and I can use them to control the lights. 
To connect the remotes to deCONZ you have to reset them by pressing the 0 button until the lights flash. Then, If you press the button with the two lights until the lights flash again, the remote is connected to deCONZ.

closed 2019-02-10 by stale[bot]

info-path, 2020-04-01:
Hello, my switch "Trust ZYCT-202" has been paired with Photoscon successfuly and can be used in a Home assistant. Unfortunately, it not listed at all in the Photoscon. I thought it was solved, but nothing was showing in the sensors/switches list. Sensors/switches from other manufacturers are listed there.