dresden-elektronik/deconz-rest-plugin #753: Ikea ct Bulbs. Sensitive timing issues. Could lead to hung bulbs, requiring a power cycle to recover.

olemr, 2018-08-26:
Tested on 2.05.35.
I have seen several times that my facade lights, 16x980lm Ikea ct bulbs falls offline when commands are sent to close in time. (to the group). Only way to get them online again is a power cycle.
Over e period of 50min at dusk, I start my lights off at yellow, low brightness.
Every minute I send a "on" command followed by a "bri" sommand 50ms later. After just a couple of nights, I have lost 2-4 bulbs on a regular basis. Things improved after I changed to just one "on" and then just "bri" commands.
Also, issuing several commands in one message to a Ikea light does not work as it should:
{ "on": true, "bri": 100,  "ct": 454 } and then { "on": true, "bri": 200,  "ct": 454 } just make the brightness go opp for a short while and it settles back to 100 again. Issuing just the "bri" command works fine.

My question is: could the gateway/rest-api be modified to 'know' it was talking to an Ikea light and properly space the commands for correct performance? I guess someone who has sniffed to Ikea Gateway traffic has the timing info required.

manup, 2018-08-26:
Interesting, can you provide the firmware version of the dropped out bulbs?

In the past there was a group command spacing in the API for other reasons, we may re-enable it if it solves the problem. But spacing might not be the real issue, on + bri results in a different command than on followed by bri, so maybe the IKEA firmware has issues with specific commands or parameters carried in them.

olemr, 2018-08-26:
I have also experienced something similar on a:
 "modelid": "TRADFRI bulb E27 CWS opal 600lm",  "swversion": "1.3.002",
When setting "xy" color values, the "bri* command has to be delayed 2s for the bulb not to hang and go offline. Sending "xy"+"bri" in one go worked as for the ct bulbs.
Also tried sending "xy" to a group og color bulbs: works fine, but going back from "xy" to "ct" does not work for groups.

ebaauw, 2018-08-26:
There’s no single ZigBee command to set both `bri` and `ct` or `xy`.   The REST API plugin breaks the single state body up into multiple ZigBee commands, _Move to level_ and _Move to color temperature_ or _Move to xy_.  I suppose it will always set `bri` before setting `ct` or `xy`, so the issue might be that the bulb needs some time to handle `ct` or `xy`.  

The colour bulb doesn’t support _Move to color temperature_.  I think they recently hacked something in the REST API plugin to translate `ct` to `xy` values and issue a _Move to xy_ instead, but that doesn’t apply to groups.

I also found the Ikea color bulb will hang if it receives invalid `xy` values, or rather `xy` values it cannot display.  Especially when setting a `y` value of 0 - I belive this causes a division by zero error in the bulb firmware, see 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/195#issuecomment-334960340

closed 2018-12-31 by stale[bot]