dresden-elektronik/deconz-rest-plugin #754: Possibility to disable homekit/homebridge

mat1990dj, 2018-08-26:
I'm using deconz together with pimatic to give pimatic access to the few zigbee devices I've got an to make pimatic the "brain" of my smart home. This means that I access the devices connected to deconz via pimatic as is where I've got all the rules.
On pimatic I've loaded the homekit/homebridge plugin so that pimatic publishes the devices on the raspberry pi port 9000 the devices.
Now when starting deconz it's also trying to get port 9000 and i've got to be very carefull on the startup order so that pimatic gets port 9000 and not deconz.
It would be great to disable homekit/homebridge.

ooii, 2018-08-27:
What about changing the port of deconz? That can be done in the config file like `ExecStart=/usr/bin/deCONZ -platform minimal --http-port=8080`.

mat1990dj, 2018-08-27:
I've already changed the port but it seems that the 9000 is always used in a fixed way, as I think that homekit protocol enforces that port usage

ebaauw, 2018-08-27:
No, the port used by homebridge is specified in `~/.homebridge/config.json`.  `deCONZ-homebridge.sh` creates this file when it's not already present and sets the port to 51826.

mat1990dj, 2018-08-27:
Then what is using port 9000?

ooii, 2018-08-27:
You only can tell. You can use either `netstat` or `lsof` (and `google`) to find which program is listening on that port.

mat1990dj, 2018-08-28:
As I've edited on the main message, I was mistaken, the conflicting port is 1900.

mat1990dj, 2018-08-28:
I'm running deCONZ on raspbian jessie, deCONZ version 2.05.23

ebaauw, 2018-08-28:
See #274 how to prevent deCONZ from using UPnP altogether.
See #83 for the issue of deCONZ not opening the port in shared mode.  I think is was corrected way before 2.05.33, but maybe it's the Jessie implementation of QT?

You could try my `upnp` utility from https://github.com/ebaauw/homebridge-hue-utils to check whether deCONZ or pimatic is the culprit.  Start one of these and run `upnp -l`.  If it reports an EADDRINUSE error, you know the one running opened the port exclusively.

mat1990dj, 2018-08-28:
Ok, so I tried the utility and gives me EADDRINUSE. I'll try upnp=0 when I get back home for now. It seems it's because of Jessie's qt

FWillems, 2018-09-08:
For me a comparable issue was solved by using

```
deCONZ --upnp=0
```

https://forum.pimatic.org/topic/4370/solved-pimatic-and-raspbee-deconz-on-same-raspberry-pi/2

mat1990dj, 2018-10-24:
Ok thanks

closed 2018-10-24 by mat1990dj