dresden-elektronik/deconz-rest-plugin #756: Aqara Motion Sensors - renaming different instances/ressources

Paalap, 2018-08-27:
My Motion sensor is reporting the values on different ids (with differenz types for light level and presence) - it is possible provide diffent names for each id, actually like it is done for the magic cube?

```
"18":{
	"config":{
		"battery":100,
		"on":true,
		"reachable":true,
		"temperature":2800,
		"tholddark":12000,
		"tholdoffset":7000
	},
	"ep":1,
	"etag":"xxx",
	"manufacturername":"LUMI",
	"modelid":"lumi.sensor_motion.aq2",
	"name":"MiMotionF2",
	"state":{
		"dark":true,
		"daylight":false,
		"lastupdated":"2018-08-27T12:53:39",
		"lightlevel":6021,
		"lux":4
	},
	"type":"ZHALightLevel",
	"uniqueid":"xxx-0400"
},
"19":{
	"config":{
		"battery":100,
		"duration":60,
		"on":true,
		"reachable":true,
		"temperature":2800
	},
	"ep":1,
	"etag":"xxx",
	"manufacturername":"LUMI",
	"modelid":"lumi.sensor_motion.aq2",
	"name":"MiMotionF2",
	"state":{
		"lastupdated":"2018-08-27T12:53:39",
		"presence":false
	},
	"type":"ZHAPresence",
	"uniqueid":"xxx-0406"
}
```

ebaauw, 2018-08-27:
Yes, through the API, just do a PUT of `/sensors/18` with a body of `{"name": "Light Level"}` and of `/sensors/19` with `{"name": "Motion"}`.

Paalap, 2018-08-27:
Thanks a lot, this helps to fix some issues in iobroker (with missed updates)

closed 2018-08-27 by Paalap

manup, 2018-08-27:
Does ioBroker use the sensor names to match devices? It would be more robust when id/uniqueid is used for this purpose instead of specific names.

Paalap, 2018-08-30:
Yes and no :)

To be more detailed: iobroker adapter is creating objects with the names used in deconz and is also saving the id as attribute. In case of the motion sensor or my example above first object "MiMotionF2" with id 18 is created but then this is updated with the id 19 :)

iobroker adapter is not considering the fact there could be different sensors with the same name and same endpoint (for both ep is 1, in case of magic cube ep is 2 and 3 - this is also the reason why magic cube is working fine, as mention in my first post)