dresden-elektronik/deconz-rest-plugin #757: Unreachable lumi.ctrl_neutral2

ebaauw, 2018-08-27:
I've got a `lumi.ctrl_neutral2` switch (end device) which cannot be reached from deCONZ, even though it's .  The switch seems stuck in a loop, where it requests the attributes of the _Time_ cluster from the coordinator, but doesn't receive any answer.  After a couple of times, it gives up, leaves and rejoins the network, connecting to a different parent.   deCONZ insists that the node is known by 0 neighbours.  Sending a command to the switch results in error 0xA7 - no clue what that means?

Below the deCONZ log (filtered on the switch's MAC address) and the Wireshark dump (filtered on the switch's NWK address):
```
Aug 27 18:58:06 pi2 deCONZ[4989]: 18:58:05:183 APS-DATA.indication srcAddr: 0x00158d000239145e, dstAddrMode: 2, profile: 0x0104, cluster: 0x000A, lqi: 231, rssi: -62
Aug 27 18:58:06 pi2 deCONZ[4989]: 18:58:05:527 add task 378160 type 21 to 0x00158D000239145E cluster 0x0004 req.id 129
Aug 27 18:58:06 pi2 deCONZ[4989]: 18:58:05:528 APS-DATA.request id: 129, addrmode: 0x03, addr: 0x00158d000239145e, profile: 0x0104, cluster: 0x0004, ep: 0x03 queue: 0 len: 4
Aug 27 18:58:24 pi2 deCONZ[4989]: 18:58:22:082 Node 0x00158D000239145E is known by 0 neighbors, last seen 16 s
Aug 27 18:59:08 pi2 deCONZ[4989]: 18:59:04:626 Node 0x00158D000239145E is known by 0 neighbors, last seen 59 s
Aug 27 18:59:08 pi2 deCONZ[4989]: 18:59:07:186 APS-DATA.indication srcAddr: 0x00158d000239145e, dstAddrMode: 2, profile: 0x0104, cluster: 0x000A, lqi: 223, rssi: -63
Aug 27 18:59:49 pi2 deCONZ[4989]: 18:59:48:571 Node 0x00158D000239145E is known by 0 neighbors, last seen 41 s
Aug 27 19:00:14 pi2 deCONZ[4989]: 19:00:09:301 APS-DATA.indication srcAddr: 0x00158d000239145e, dstAddrMode: 2, profile: 0x0104, cluster: 0x000A, lqi: 223, rssi: -63
Aug 27 19:00:39 pi2 deCONZ[4989]: 19:00:36:883 Node 0x00158D000239145E is known by 0 neighbors, last seen 27 s
Aug 27 19:00:39 pi2 deCONZ[4989]: 19:00:38:974 APS-DATA.indication srcAddr: 0x00158d000239145e, dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 191, rssi: -67
Aug 27 19:00:39 pi2 deCONZ[4989]: 19:00:38:975 DeviceAnnce of LightNode: 0x00158d000239145e Permit Join: 0
Aug 27 19:00:39 pi2 deCONZ[4989]: 19:00:38:975 DeviceAnnce of SensorNode: 0x00158D000239145E [1]
Aug 27 19:00:39 pi2 deCONZ[4989]: 19:00:38:975 DeviceAnnce of SensorNode: 0x00158D000239145E [1]
Aug 27 19:00:39 pi2 deCONZ[4989]: 19:00:38:975 device announce 0x00158D000239145E (0x7F15) mac capabilities 0x84
Aug 27 19:00:39 pi2 deCONZ[4989]: 19:00:38:975 set fast probe address to 0x00158D000239145E (0x7F15)
Aug 27 19:00:46 pi2 deCONZ[4989]: 19:00:45:276 add task 378937 type 21 to 0x00158D000239145E cluster 0x0004 req.id 129
Aug 27 19:00:46 pi2 deCONZ[4989]: 19:00:45:276 APS-DATA.request id: 129, addrmode: 0x03, addr: 0x00158d000239145e, profile: 0x0104, cluster: 0x0004, ep: 0x02 queue: 2 len: 4
Aug 27 19:00:46 pi2 deCONZ[4989]: 19:00:46:181 add task 378941 type 21 to 0x00158D000239145E cluster 0x0004 req.id 133
Aug 27 19:00:46 pi2 deCONZ[4989]: 19:00:46:181 APS-DATA.request id: 133, addrmode: 0x03, addr: 0x00158d000239145e, profile: 0x0104, cluster: 0x0004, ep: 0x03 queue: 2 len: 4
Aug 27 19:01:13 pi2 deCONZ[4989]: 19:01:11:215 APS-DATA.indication srcAddr: 0x00158d000239145e, dstAddrMode: 2, profile: 0x0104, cluster: 0x000A, lqi: 207, rssi: -65
Aug 27 19:01:25 pi2 deCONZ[4989]: 19:01:23:419 Node 0x00158D000239145E is known by 0 neighbors, last seen 12 s
Aug 27 19:02:09 pi2 deCONZ[4989]: 19:02:07:354 Node 0x00158D000239145E is known by 0 neighbors, last seen 56 s
Aug 27 19:02:16 pi2 deCONZ[4989]: 19:02:13:228 APS-DATA.indication srcAddr: 0x00158d000239145e, dstAddrMode: 2, profile: 0x0104, cluster: 0x000A, lqi: 199, rssi: -66
Aug 27 19:02:24 pi2 deCONZ[4989]: 19:02:24:107 add task 379420 type 14 to 0x00158D000239145E cluster 0x0006 req.id 121
Aug 27 19:02:25 pi2 deCONZ[4989]: 19:02:24:109 APS-DATA.request id: 121, addrmode: 0x03, addr: 0x00158d000239145e, profile: 0x0104, cluster: 0x0006, ep: 0x02 queue: 2 len: 3
Aug 27 19:02:30 pi2 deCONZ[4989]: 19:02:29:052 0x00158D000239145E error APSDE-DATA.confirm: 0xA7 on task
Aug 27 19:02:51 pi2 deCONZ[4989]: 19:02:51:154 Node 0x00158D000239145E is known by 0 neighbors, last seen 37 s
Aug 27 19:03:17 pi2 deCONZ[4989]: 19:03:15:254 APS-DATA.indication srcAddr: 0x00158d000239145e, dstAddrMode: 2, profile: 0x0104, cluster: 0x000A, lqi: 191, rssi: -67
Aug 27 19:03:41 pi2 deCONZ[4989]: 19:03:37:882 Node 0x00158D000239145E is known by 0 neighbors, last seen 22 s
Aug 27 19:04:20 pi2 deCONZ[4989]: 19:04:17:282 APS-DATA.indication srcAddr: 0x00158d000239145e, dstAddrMode: 2, profile: 0x0104, cluster: 0x000A, lqi: 255, rssi: -53
Aug 27 19:04:28 pi2 deCONZ[4989]: 19:04:24:650 Node 0x00158D000239145E is known by 0 neighbors, last seen 7 s
Aug 27 19:04:40 pi2 deCONZ[4989]: 19:04:39:276 add task 380084 type 21 to 0x00158D000239145E cluster 0x0004 req.id 214
Aug 27 19:04:40 pi2 deCONZ[4989]: 19:04:39:277 APS-DATA.request id: 214, addrmode: 0x03, addr: 0x00158d000239145e, profile: 0x0104, cluster: 0x0004, ep: 0x02 queue: 1 len: 4
Aug 27 19:05:14 pi2 deCONZ[4989]: 19:05:08:650 Node 0x00158D000239145E is known by 0 neighbors, last seen 51 s
Aug 27 19:05:54 pi2 deCONZ[4989]: 19:05:52:315 Node 0x00158D000239145E is known by 0 neighbors, last seen 95 s
Aug 27 19:06:03 pi2 deCONZ[4989]: 19:06:01:408 APS-DATA.indication srcAddr: 0x00158d000239145e, dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 191, rssi: -67
Aug 27 19:06:03 pi2 deCONZ[4989]: 19:06:01:409 DeviceAnnce of LightNode: 0x00158d000239145e Permit Join: 0
Aug 27 19:06:03 pi2 deCONZ[4989]: 19:06:01:409 DeviceAnnce of SensorNode: 0x00158D000239145E [1]
Aug 27 19:06:03 pi2 deCONZ[4989]: 19:06:01:409 DeviceAnnce of SensorNode: 0x00158D000239145E [1]
Aug 27 19:06:03 pi2 deCONZ[4989]: 19:06:01:410 device announce 0x00158D000239145E (0x7F15) mac capabilities 0x84
Aug 27 19:06:03 pi2 deCONZ[4989]: 19:06:01:410 set fast probe address to 0x00158D000239145E (0x7F15)
Aug 27 19:06:03 pi2 deCONZ[4989]: 19:06:03:277 read attributes of 0x00158D000239145E cluster: 0x0000: [ 19:06:03:277 0x4000 19:06:03:277 ]
Aug 27 19:06:03 pi2 deCONZ[4989]: 19:06:03:277 add task 380490 type 19 to 0x00158D000239145E cluster 0x0000 req.id 113
Aug 27 19:06:03 pi2 deCONZ[4989]: 19:06:03:277 add task 380491 type 21 to 0x00158D000239145E cluster 0x0004 req.id 114
Aug 27 19:06:03 pi2 deCONZ[4989]: 19:06:03:277 APS-DATA.request id: 113, addrmode: 0x03, addr: 0x00158d000239145e, profile: 0x0104, cluster: 0x0000, ep: 0x02 queue: 0 len: 5
Aug 27 19:06:03 pi2 deCONZ[4989]: 19:06:03:327 delay sending request 114 dt 0 ms to 0x00158D000239145E, cluster 0x0004
Aug 27 19:06:03 pi2 deCONZ[4989]: 19:06:03:327 delay sending request 114 cluster 0x0004 to 0x00158d000239145e
Aug 27 19:06:06 pi2 deCONZ[4989]: 19:06:03:403 APS-DATA.request id: 114, addrmode: 0x03, addr: 0x00158d000239145e, profile: 0x0104, cluster: 0x0004, ep: 0x02 queue: 1 len: 4
Aug 27 19:06:06 pi2 deCONZ[4989]: 19:06:04:027 read attributes of 0x00158D000239145E cluster: 0x0000: [ 19:06:04:027 0x4000 19:06:04:027 ]
Aug 27 19:06:06 pi2 deCONZ[4989]: 19:06:04:027 add task 380495 type 19 to 0x00158D000239145E cluster 0x0000 req.id 118
Aug 27 19:06:06 pi2 deCONZ[4989]: 19:06:04:027 add task 380496 type 21 to 0x00158D000239145E cluster 0x0004 req.id 119
Aug 27 19:06:06 pi2 deCONZ[4989]: 19:06:04:027 APS-DATA.request id: 118, addrmode: 0x03, addr: 0x00158d000239145e, profile: 0x0104, cluster: 0x0000, ep: 0x03 queue: 0 len: 5
Aug 27 19:06:06 pi2 deCONZ[4989]: 19:06:04:029 delay sending request 119 dt 0 ms to 0x00158D000239145E, cluster 0x0004
Aug 27 19:06:06 pi2 deCONZ[4989]: 19:06:04:029 delay sending request 119 cluster 0x0004 to 0x00158d000239145e
Aug 27 19:06:06 pi2 deCONZ[4989]: 19:06:04:126 delay sending request 119 dt 0 ms to 0x00158D000239145E, cluster 0x0004
Aug 27 19:06:06 pi2 deCONZ[4989]: 19:06:04:126 delay sending request 119 cluster 0x0004 to 0x00158d000239145e
Aug 27 19:06:06 pi2 deCONZ[4989]: 19:06:04:161 APS-DATA.request id: 119, addrmode: 0x03, addr: 0x00158d000239145e, profile: 0x0104, cluster: 0x0004, ep: 0x03 queue: 1 len: 4
```
[wireshark.pcapng.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2324858/wireshark.pcapng.zip)

ebaauw, 2018-08-27:
But after a while, it leaves and rejoins the network again, and becomes unreachable again?!

Double-checked my other `lumi.ctrl_neutral2`: it also sends the _Read attributes_ command for the _Time_ cluster to the coordinator, and it also leaves/re-joins, finding a new parent, but deCONZ reports that it's know by 1 neighbour.

ebaauw, 2018-08-29:
Today, my light nodes began turning red, starting, as always, with the Hue Blooms and (old model) Hue LED strips.  This still happens once every other month or so.  As usual, simply power cycling the red nodes doesn't help; only a full power cycle of my entire network remedied the situation.  Both `lumi.ctrl_neutral2` switches are now again fully operational.  They're still asking the coordinator for the time, and they're still leaving and re-joining the network, but remain reachable.

@manup, I suspect there might still be something wrong with the routing in firmware 261f0500 that's causing the unresponsive nodes, but I have no clue how to even start analysing this.  I never saw these unrecoverable red nodes, until the firmware was changed to handle IKEA meshing.  I now have 89 nodes in my production network (1 coordinator, 47 routers).  I haven't seen this issue on my test network.

manup, 2018-08-29:
They might use the a non responsive time cluster as indication to rejoin the network, same as Philips end-devices do when they don't receive zcl default response for their reports. A solution might be to just implement the time server on our side (or at least the responses which they expect).

In the last days I was quite busy to implement full nodes restore right after deCONZ starts, this is nearly finished and works quite well in my test networks (34, 80 and 105 nodes), hope to get the next release out this week. 

Also I found a serious bug in the firmware (all versions) which will be fixed in 26220500, it caused the ZDO/ZDP module of the firmware to be blocked until next restart. As for the IKEA/OSRAM meshing I'm tempted to disable the many-to-one route record feature, it seems to cause more harm than good. All the other optimizations within deCONZ of the past releases should already provide a stable network, so it should be safe to disable it and just rely on good'ol mesh routing.

ebaauw, 2018-08-30:
> Both `lumi.ctrl_neutral2` switches are now again fully operational.

That didn't last too long.  Now seeing this error again:
```
Aug 30 09:31:51 pi2 deCONZ[6429]: 09:31:48:856 0x00158D000239145E error APSDE-DATA.confirm: 0xA7 on task
```
@manup, what's error 0xA7 in English?  Would that be indicative of the ZDO/ZDP module of the firmware being blocked?

manup, 2018-08-30:
`0xA7` is no APS ACK received.
Since the switch often rejoins it might get a new NWK address and thus this might be the reason for the 0xA7.

ebaauw, 2018-08-30:
Thanks.  I’ll see if I can cook up something in the REST API plugin to answer the time request.

ebaauw, 2018-08-30:
The NWK address doesn’t seem to change, but the switch rejoins to a different parent.

manup, 2018-08-30:
Interesting, I guess since the route changes it will need a few attempts until the gateway can reach the switch again. 0xA7 and somethings followed by 0xD0  (No Route), then route discovery restarts.

Keeping the switch happy via time cluster should be a good solution to prevent that.

ebaauw, 2018-08-30:
Hacked together a very basic _Time_ server, only responding to the _Read Attributes_ command for the _Time_ attribute (0x0000).  As far as I can tell from WireShark, it works correctly, responding with a _Read Attributes Response_ containing the current time (see frames below).

Once of my switches seems to like it: although it keeps spamming the coordinator for the current time, it no longer leaves and re-joins the network.  I'm in a brilliant catch 22 situation for the second switch: it seems the _Read Attributes Response_ doesn't reach the switch (neither do other commands), so it continues to leave/re-join.  I'll try another power cycle of the RaspBee.
 
Here's the request from the switch:
```
Frame 25777: 124 bytes on wire (992 bits), 124 bytes captured (992 bits) on interface 0
Ethernet II, Src: Raspberr_3b:e8:df (b8:27:eb:3b:e8:df), Dst: Apple_ca:3b:53 (a4:5e:60:ca:3b:53)
Internet Protocol Version 4, Src: 192.168.76.81, Dst: 192.168.76.41
User Datagram Protocol, Src Port: 53808, Dst Port: 17754
ZigBee Encapsulation Protocol, Channel: 25, Length: 50
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0xa72e
ZigBee Network Layer Data, Dst: 0x0000, Src: 0xdc9f
    Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
    Destination: 0x0000
    Source: 0xdc9f
    Radius: 29
    Sequence Number: 112
    [Extended Source: Jennic_00:02:4b:d5:7e (00:15:8d:00:02:4b:d5:7e)]
    [Origin: 643]
    ZigBee Security Header
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Time (0x000a)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 66
ZigBee Cluster Library Frame, Command: Read Attributes, Seq: 216
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 216
    Command: Read Attributes (0x00)
    Attribute: Time (0x0000)
```
And here's my response:
```
Frame 25779: 130 bytes on wire (1040 bits), 130 bytes captured (1040 bits) on interface 0
Ethernet II, Src: Raspberr_3b:e8:df (b8:27:eb:3b:e8:df), Dst: Apple_ca:3b:53 (a4:5e:60:ca:3b:53)
Internet Protocol Version 4, Src: 192.168.76.81, Dst: 192.168.76.41
User Datagram Protocol, Src Port: 53808, Dst Port: 17754
ZigBee Encapsulation Protocol, Channel: 25, Length: 56
IEEE 802.15.4 Data, Dst: 0x3ddd, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xdc9f, Src: 0x0000
    Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
    Destination: 0xdc9f
    Source: 0x0000
    Radius: 10
    Sequence Number: 9
    [Extended Source: Dresden-_ff:ff:01:02:56 (00:21:2e:ff:ff:01:02:56)]
    [Origin: 74]
    ZigBee Security Header
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Time (0x000a)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 174
ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 216
    Frame Control Field: Profile-wide (0x18)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 1... = Direction: Client
        ...1 .... = Disable Default Response: True
    Sequence Number: 216
    Command: Read Attributes Response (0x01)
    Status Record, UTC
        Attribute: Time (0x0000)
        Status: Success (0x00)
        Data Type: UTC Time (0xe2)
        UTC: Aug 30, 2018 22:21:58.000000000 CEST
```
Looks like WireShark courteously translates the UTC value to local time.

manup, 2018-09-01:
Cool thanks, looks very good to me, the PR is merged and available in 2.05.37. You may try the 0x26220500 firmware included in the release, will bring hopefully more stable routing.

ebaauw, 2018-09-02:
Now running v2.05.37 and 0x26220500 in test and production.  Some observations:
- Indeed, after restart, the GUI immediately shows all 89 nodes, and starts drawing the connection lines;
- All `/lights` resources are available immediately - I think I can remove the `lightlist` workaround from homebridge-hue;
- `state.reachable` is true immediately after startup for all the `/lights` resources - I'd expect this to be false until deCONZ has actually communicated with the light.
- Same for `config.reachable` for `/sensors` resources (these used to available immediately after startup, but only became reachable as they were discovered);
- The GUI lost the fields in the _Node info_ panel after the first start, but these were re-populeted after next restarts.  However, the name of my RaspBee is lost on each restart (both in the GUI node as well as in the _Node info_ panel).  Also the name of the XBee-s is lost;
- The positions of the XBee nodes is lost on the first start, but saved at next restarts.
- Firmware wouldn't update from the old web app, even though GUI showed the firmware update bar and I unplugged all USB devices (ConBee and XBee-s).  No issue updating through GCFFlasher_internal.

ebaauw, 2018-09-02:
After restart, deCONZ has a funny state for the `lumi.ctrl_neutral2` switches.  deCONZ reports them as unavailable, until I manually read the _OnOff_ cluster attributes from the GUI.  This was also the case in v2.05.35:
```
$ ph get /lights/423
{
  "etag": "3afa8b361233e55d9a07887e8f6ae0ac",
  "hascolor": false,
  "manufacturername": "LUMI",
  "modelid": "lumi.ctrl_neutral2",
  "name": "Living Room Fan 2",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swversion": "11-11-2016",
  "type": "Smart plug",
  "uniqueid": "00:15:8d:00:02:4b:d5:7e-02"
}
$ ph put /lights/423/state '{"on": true}'
ph put: error: 3 resource, /lights/423, not available
```
I think this is due to the switch reporting _Receiver On When Idle_ as false, even though it's mains powered.
[image]
However, the _Basic_ cluster reports it's on DC power:
[image]
I can't wait for deCONZ to read the _Power descriptor_ again.

ebaauw, 2018-09-02:
The `lumi.ctrl_neutral2` switches don't seem to react to group commands (nor to a `/groups/0` broadcast), even though they seem to support _Groups_ cluster commands on both outlet endpoints.
```
Sep 02 18:23:04 pi2 deCONZ[312]: 18:22:59:258 add task 4777 type 21 to 0x00158D00024BD57E cluster 0x0004 req.id 125
Sep 02 18:23:04 pi2 deCONZ[312]: 18:22:59:258 APS-DATA.request id: 125, addrmode: 0x03, addr: 0x00158d00024bd57e, profile: 0x0104, cluster: 0x0004, ep: 0x01 -> 0x02 queue: 2 len: 4 tx.options 0x00
Sep 02 18:23:04 pi2 deCONZ[312]: 18:22:59:995 APS-DATA.indication srcAddr: 0x00158d00024bd57e, dstAddrMode: 2, profile: 0x0104, cluster: 0x0004, lqi: 215, rssi: -64
Sep 02 18:23:04 pi2 deCONZ[312]: 18:22:59:995 verified group capacity: 7 and group count: 1 of LightNode 0x00158d00024bd57e
Sep 02 18:23:04 pi2 deCONZ[312]: 18:22:59:995 0x00158d00024bd57e found group 0x0003
...
Sep 02 18:28:04 pi2 deCONZ[312]: 18:28:02:505 add task 6263 type 21 to 0x00158D00024BD57E cluster 0x0004 req.id 159
Sep 02 18:28:04 pi2 deCONZ[312]: 18:28:02:552 APS-DATA.request id: 159, addrmode: 0x03, addr: 0x00158d00024bd57e, profile: 0x0104, cluster: 0x0004, ep: 0x01 -> 0x03 queue: 2 len: 4 tx.options 0x00
Sep 02 18:28:04 pi2 deCONZ[312]: 18:28:03:701 APS-DATA.indication srcAddr: 0x00158d00024bd57e, dstAddrMode: 2, profile: 0x0104, cluster: 0x0004, lqi: 231, rssi: -62
Sep 02 18:28:04 pi2 deCONZ[312]: 18:28:03:701 verified group capacity: 8 and group count: 0 of LightNode 0x00158d00024bd57e
```

BTW, is it possible to add the source endpoint to the `APS-DATA.indication` messages (I assume from the deCONZ core program)?

manup, 2018-09-02:
Thanks for the review, most of the issues should hopefully be fixed soon.

> * state.reachable is true immediately after startup for all the /lights resources - I'd expect this to be false until deCONZ has actually communicated with the light.

Either this or the last known state is restored, so after a deCONZ restart everything goes back to normal quickly. Currently this information is not in the database, it's already planned but didn't make it into this release :)

> * The GUI lost the fields in the Node info panel after the first start, but these were re-populeted after next restarts. However, the name of my RaspBee is lost on each restart (both in the GUI node as well as in the Node info panel). Also the name of the XBee-s is lost;

Confirmed, should be fixed in the next release. The XBees gui position did fall through since they don't have references in the zll.db. I'll add an exception.

The names of XBee and ConBee is stored in user-descriptor which is not yet stored in the zll.db.

Behind the scenes a cleanup of the session.default file happened, cached devices which have no references in zll.db were removed.

> * Firmware wouldn't update from the old web app, even though GUI showed the firmware update bar and I unplugged all USB devices (ConBee and XBee-s). No issue updating through GCFFlasher_internal.

Hmm, bad this should not happen I'll do more testing on various machines.

manup, 2018-09-02:
> BTW, is it possible to add the source endpoint to the APS-DATA.indication messages (I assume from the deCONZ core program)?

Sure, I'll add it to the debug prints.

ebaauw, 2018-09-07:
Looks like they're working (now on v2.05.38 and firmware 0x26230500), except for https://github.com/dresden-elektronik/deconz-rest-plugin/issues/757#issuecomment-417943550:
> After restart, deCONZ has a funny state for the lumi.ctrl_neutral2 switches. deCONZ reports them as unavailable, until I manually read the OnOff cluster attributes from the GUI. This was also the case in v2.05.35

@manup, what's your take on this?