dresden-elektronik/deconz-rest-plugin #758: Phoscon Switch Editor broken in v2.05.35

oscar-b, 2018-08-28:
The switch editor worked (ish) in v2.05.34, enough to configure my Hue Dimmer Switch, but after upgrading to v2.05.35 it seems completely broken.

When I load the switch editor, the old key assignments aren't there. And if I add a new one, it doesn't display in the list of key assignments.

The switch, however, still works as it did before the upgrade.

manup, 2018-08-28:
I figure it's the mixed browser cached version of the site. It should work after "some" time or when the cache is cleared/updated.

In the next version we'll fix the cache refresh after an update.

oscar-b, 2018-08-29:
I doubt that's the problem, I've cleared all caches (dev tools, application, clear storage). I also run without cache when dev tools is open (which seem to cause the Qt web server to freak out due to the amount of requests, but that's another issue).

manup, 2018-08-29:
Hmm strange, do the browser dev tools show any error? Disabled cache shouldn't have a negative effect, in fact we're developing the App in this way (mostly in Chrome Dev Tools and Safari/Firefox for testing).

We'll try to reproduce the problem with the Hue Dimmer switch and update from .34 to .35.

oscar-b, 2018-08-29:
At some point I got this message, not sure if it's related or not.

``` TODO: Schaltergruppe mit Lichtern verursacht Fehler, wenn Licht in Regel vorkommt```

manup, 2018-09-01:
Found the issue should work again in 2.05.37.

manup, 2018-09-05:
Close the Issue for now.

closed 2018-09-05 by manup

oscar-b, 2018-09-05:
@manup Will test it as soon as the Docker image is updated

Surfer2010, 2020-08-26:
Seems like this is still a problem.
Newest Docker Image (2.05.80 / 14.8.2020)
I have a Switch (Osram Lightyfy 4 Button Switch) in the "switches" menu, but no available switch in the switch editor in my Room

Mimiix, 2020-08-26:
@Surfer2010 This issue is _minus 2 days_ 2 years old. Please open a new user question issue.