dresden-elektronik/deconz-rest-plugin #759: Gateway not discovered by Alexa

ooii, 2018-08-28:
I read on another issue that deconz works fine with Alexa since several months. But in my case, I cannot discover it with Alexa. I enable the Hue skill then open the gateway and finally start the discovery. And I never find it. 
Is there something I’m doing wrong?
Thanks.

manup, 2018-08-28:
It doesn't work with the Hue skill, only with the build in Echo Hue support, normally you need to allow to connect apps in the *Phoscon App > Menu > Gateway > Settings* and then via voice command ask Alexa "discover devices".

ooii, 2018-08-28:
That’s exactly what I did. When I start detection, it says that I should enable the associated connected home skill (translated from French) if not already enabled and it starts discovering for up to 20 seconds then fails.

ooii, 2018-08-29:
Btw, I'm using an Echo Dot while you're mentioning an Echo, is there any difference?

manup, 2018-08-29:
Should also work we had testet it a while ago, but it was important not to use (or even install?) the Hue skill and just use the plain Alexa discover devices method.

ooii, 2018-08-29:
Just ran a test after disabling the Hue skill and still not able to discover my bridge.
Is there anything I can test to figure out what happening? Is there any issue with homebridge (-hue) running at the same time (I don't think so but you never know).

manup, 2018-09-02:
Just did a test with an Echo Dot and it worked.

* deCONZ runs on port 80, seems to be important since the Echo tries to connect to 80 only
* No Hue skill installed
* in the Phoscon App enable app authentication for 60 seconds
* start search for devices in Echo app smart home devices section

Is your deCONZ installation running on port 80?

ooii, 2018-09-03:
Unfortunately, that did not fix the issue. deConz is running now on port 80 but Alexa is still unable to discover it. 
Note that once, I was below the step of searching for 20 seconds and I was asked to switch on my switches and lamps, then an error occured in Alexa, and I was back to unsuccessful 20 seconds discovery.

manup, 2018-09-04:
Other than port 80 the UDP port 1900 is used for SSDP multicast (or UPnP) which will be used by the Echo to detect the machine running deCONZ. It can do this only when they both are in the same network with no network switches in between. Also deCONZ must be allowed to bind to shared port 1900, which other services running on the same port might prevent.

mopby, 2018-09-08:
Hello, I have the same problem. Alexa can't find any devices. Deconz is running on port 80. How can I figure out if UDP port 1900 is used?

mopby, 2018-09-08:
Thanks for the tip, uPnP Analyzer shows

NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=100
LOCATION: http://127.0.0.1:80/description.xml
SERVER: FreeRTOS/7.4.2, UPnP/1.0, IpBridge/1.8.0
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:d6a5efe7-3653-4212-b551-75bf4cd7bead::upnp:rootdevice
GWID.phoscon.de: 00212EFFFF022B32
hue-bridgeid: 00212EFFFF022B32

everything seems fine. Im not running any network switch, the raspberry (running deconz) is connected to my wifi router / modem (fritzbox) directly via ethernet. 

Unfortunately my echo dot still doesn't find any devices.

ebaauw, 2018-09-08:
Did you turn off IGMP snooping on your router?

ebaauw, 2018-09-08:
Oh, Fritz!Box doesn’t provide that setting.  Better connect everything to a switch and use only one port on the Fritz.  See 
https://github.com/nfarina/homebridge/issues/1801#issuecomment-417220526

mopby, 2018-09-08:
Thanks for your quick response, I'll order a switch an follow your instructions. I'll report back :)

ebaauw, 2018-09-08:
No guarantees this will solve the Alexa issue.  We've just seen that routers with IGMP snooping enabled try to be too smart when it comes to handling multicast traffic (Bonjour, UPnP/SSDP).  As a result, multicast traffic sometimes isn't delivered to the appropriate port, effectively making devices invisible for discovery by clients (even though on the server side, everything seems to be in order).  And the Fritz!Box seems to do IGMP snooping, with no option to disable it.

Before ordering a switch, you might want to try UPnP analyser on a host connected to another port on the Fritz!Box than the deCONZ server, to confirm that it (sometimes) doesn't receive the SSDP packets from deCONZ.  Unless, of course, you planned on getting a switch anyways.

ooii, 2018-09-08:
@mopby where are you running uPnP analyzer? On the raspberry which is running deconz? 
It's weird to see the service available at `LOCATION: http://127.0.0.1:80/description.xml` which is the localhost address. 

Thank you @ebaauw for the clarification about IGMP snooping.

mopby, 2018-09-08:
ok, so i would need a new wifi-router as well since the echo dot is connected to the fritzboxs wifi at the moment. Thank you ebaauw for pointing this out.

Just for testing: I unplugged all devices but the raspberry. Still not able to discover my devices.

I ran the uPnP analyzer off my mac. I wondered about the address being localhost as well. I found some old posts about this being fixed. 

I ran ha-bridge a while ago (from another machine) without any discovery problems (also connected via ethernet to my fritzbox).

ooii, 2018-09-08:
It seems you have an issue with that. The provided url cannot allow Alexa to get the xml file correctly.

mopby, 2018-09-08:
do you have an idea how to fix this?

ebaauw, 2018-09-08:
> It's weird to see the service available at LOCATION: http://127.0.0.1:80/description.xml which is the localhost address.
> I found some old posts about this being fixed.

What version of deCONZ are you on?  v2.05.38 shows the Raspberry's IP address.

mopby, 2018-09-08:
Im on the newest version (2.05.38). Already did reinstall and check if everythings alright. Since updating I also noticed that under "settings" -> "gateway" the firmware number is 00000000. When I upgrade the firmware everythings all right. But after a reboot the number is 00000000 again and I can click the upgrade button again.

I tried reinstalling deconz, same behaviour.

ebaauw, 2018-09-08:
Are you running natively on Raspbian or in a Docker container?  If natively, what version of Raspbian?

> I ran ha-bridge a while ago (from another machine) without any discovery problems (also connected via ethernet to my fritzbox)

I don't know how strict the Echo Dot or ha-bridge check that they found deCONZ (or a Hue bridge).  homebridge-hue is happy with `hue-bridgeid` and `GWID.phoscon.de` and completely ignores `LOCATION`.

mopby, 2018-09-10:
Hello, i experimented with my deconz installation. I moved deconz to a virtual ubuntu installation (running on my proxmox) and now alexa finds all devices. Also the uPnP advertises as it should.

NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=100
LOCATION: http://192.168.178.30:80/description.xml
SERVER: FreeRTOS/7.4.2, UPnP/1.0, IpBridge/1.8.0
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:d6a5efe7-3653-4212-b551-75bf4cd7bead::upnp:rootdevice
GWID.phoscon.de: 00212EFFFF022B32
hue-bridgeid: 00212EFFFF022B32

Stillt don't know whats wrong with my raspberry, but I'm very happy now with everything working as expected. Not a bug, just a fault of my system I guess. Thanks for your help and sorry for not being able to provide help for others since I didn't find the source of my problem

Spitzbua, 2018-10-14:
I have the same problem with my configuration. Alexa on RPI can't detect the Phoscon gateway.

Followed those steps, but no success:

- deCONZ runs on port 80, seems to be important since the Echo tries to connect to 80 only
- No Hue skill installed
- in the Phoscon App enable app authentication for 60 seconds
- start search for devices in Echo app smart home devices section

UPNP is running and file can be received through http://192.168.0.4:80/description.xml:

`pi@raspalexa:~ $ gssdp-discover -i eth0 --timeout=3
Using network interface eth0
Scanning for all resources
Showing "available" messages
resource available
  USN:      uuid:fdacb23f-a3f0-4e12-8d5b-303705579304::upnp:rootdevice
  Location: http://192.168.0.4:80/description.xml
resource available
  USN:      uuid:fdacb23f-a3f0-4e12-8d5b-303705579304
  Location: http://192.168.0.4:80/description.xml
resource available`

Detection worked with HUE app and All4Hue but not with Alexa. I have also tried to set both on the same switch, but no success. Any other ideas what to do? The gateway will be detected by several other apps....

saurabh984, 2018-10-15:
Same here. Using v2.05.40

Had to reset all my Echo devices and it can't discover deCONZ anymore. 

I can see deCONZ in using uPnP discovery tools.

tiwo85, 2018-10-26:
Same here. U sind 2.05.42

saurabh984, 2018-10-28:
I can confirm @manup fix https://github.com/dresden-elektronik/deconz-rest-plugin/commit/2c0c974165a9b4f08d4b7e31d931c6fae7bfc74a has fixed Alexa discovery issue.

Thank you :)

Spitzbua, 2018-11-03:
@saurabh984 , I have now updated to 2.05.45, altough the firmware update did not work over GCFFlasher or over the WebApp.  What firmware are you using?

I am still facing the discovery issue with Alexa:

Version 2.05.45 / 2.11.2018
Firmware 26240500

fretmeat, 2018-11-08:
@Spitzbua,
Same here - Alexa/Echo can't find the fixtures etc. in 2.05.46.
Used to work flawlessly until I had to re-setup everything.

My first guess is that the recent Alexa app/interface overhaul changed the way the (fake) HUE bridge is contacted by the echo - as 3rd party apps still seem work.  

Anyhow - I think the Echo stopped "seeing" the fixtures once I update from 2.05.34 to 2.05.44 a couple of days ago - so second guess is that sometime in between something changed the behaviour.

Just tried HUE essentials on Android and it works.
It askes for the password of the gateway, which alexa doesn't 
(I think the push-button solution used to work before)
//(well-did I just tell some app my password, I guess so.)

saurabh984, 2018-11-08:
@Spitzbua 0x262e0500

I moved to a ubuntu VM on esxi. It didn't work for me on Raspberry Pi. I don't this is to do with changes to API. I think it might be a routing issue or timing. UPnP Analyser showed a response from Raspberry Pi after 20 seconds. Alexa tries the UPnP discovery for a lot less than that.

KodeCR, 2018-11-08:
Looks like I broke Alexa discovery in 2.05.44 (see #902), Alexa apparently wants the manufacturer in description.xml to be Philips. Try updating /usr/share/deCONZ/webapp/description_in.xml with the latest one from github (https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/description_in.xml), or manually change "<manufacturer>dresden elektronik</manufacturer>" to "<manufacturer>Royal Philips Electronics</manufacturer>".

On my Raspberry Pi it definitely works, I used the latest version of the app, "Devices", "+", "Add device", "Hue", "Hue bridge v1", "Authenticate App" in Phoson, "Discover devices". Note that it is important that the Hue skill is not enabled, otherwise you can't choose the v1 bridge.

ooii, 2018-11-09:
Thanks @KodeCR but it is not working here. Changed the description.xml file and restarted deconz. The xml exposed throuhg uPnP shows Royal Philips Electronics now instead of Dresden elektronic but still not able to discover it.

saurabh984, 2018-11-10:
That fixed some new lights I just added too! The description_in.xml doesn't get replaced by the deb package for some reason. It had the incorrect manufacturer name even after the latest package installed.

ooii, 2018-11-13:
I was reading some instructions about integration of Alexa with [Home Assistant](https://www.home-assistant.io) and found that: `Warning: If you have a lot of devices (more than 49 total across all exposed domains)`. Since I have more than 49 devices in my deConz setup, I think that may be the cause of my Alexa not discovering my devices.

giancarlolamedica, 2018-11-18:
No way to find bridge from alexa android or ios app. The phone is connected directly to wifi of raspberry configured as access point. Raspberry is running raspbian jessie, deconz 2.05.47 with firmware 262F0500, i have 23 nodes on zigbee network. SSDPTester (an android app) receive correctly the uPnp packet after few seconds.

18 nov 2018 14:15:39: Packet received from 192.168.1.7:1900 :
NOTIFY * HTTP/1.1\r\n
HOST: 239.255.255.250:1900\r\n
CACHE-CONTROL: max-age=100\r\n
LOCATION: http://192.168.1.7:80/description.xml\r\n
SERVER: Linux/3.14.0 UPnP/1.0 IpBridge/1.26.0\r\n
GWID.phoscon.de: ***censored***\r\n
hue-bridgeid: ***censored***\r\n
NTS: ssdp:alive\r\n
NT: urn:schemas-upnp-org:device:basic:1\r\n
USN: uuid:***censored***::urn:schemas-upnp-org:device:basic:1\r\n
\r\n

this is the content of http://192.168.1.7:80/description.xml

<?xml version="1.0" encoding="UTF-8"?>
<root xmlns="urn:schemas-upnp-org:device-1-0">
   <specVersion>
      <major>1</major>
      <minor>0</minor>
   </specVersion>
   <URLBase>http://192.168.1.7:80/</URLBase>
   <device>
      <deviceType>urn:schemas-upnp-org:device:Basic:1</deviceType>
      <friendlyName>Phoscon-GW (192.168.1.7)</friendlyName>
      <manufacturer>Royal Philips Electronics</manufacturer>
      <manufacturerURL>http://www.dresden-elektronik.de</manufacturerURL>
      <modelDescription>Philips hue compatible Personal Wireless Lighting</modelDescription>
      <modelName>Philips hue bridge 2015</modelName>
      <modelNumber>BSB002</modelNumber>
      <modelURL>http://www.dresden-elektronik.de</modelURL>
      <serialNumber>***censored***</serialNumber>
      <UDN>uuid:***censored***</UDN>
      <presentationURL>index.html</presentationURL>
      <iconList>
         <icon>
            <mimetype>image/png</mimetype>
            <height>48</height>
            <width>48</width>
            <depth>24</depth>
            <url>hue_logo_0.png</url>
         </icon>
      </iconList>
   </device>
</root>

Scan device from alexa app (hue bridge v1 with no skill enabled and 3rd party apps authentication enabled from gateway) fail after 20 seconds (tried several times). I don't have a physical echo device at the moment but i think it's ininfluent.
I tried also to remove the router (attached to raspberry via ethernet) so that the deconz is the only source of ssdp packets. If sniff packets on wlan interface of raspberry while scanning device from alexa don't see ssdp search request but a dhcp discover and i see ssdp notification from deconz with destination 239.255.255.250

NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=100
LOCATION: http://192.168.2.8:80/description.xml
SERVER: Linux/3.14.0 UPnP/1.0 IpBridge/1.26.0
GWID.phoscon.de: ***censored***
hue-bridgeid: ***censored***
NTS: ssdp:alive
NT: urn:schemas-upnp-org:device:basic:1
USN: uuid:***censored***::urn:schemas-upnp-org:device:basic:1

and xml with this network configuration

<?xml version="1.0" encoding="UTF-8"?>
<root xmlns="urn:schemas-upnp-org:device-1-0">
   <specVersion>
      <major>1</major>
      <minor>0</minor>
   </specVersion>
   <URLBase>http://192.168.2.8:80/</URLBase>
   <device>
      <deviceType>urn:schemas-upnp-org:device:Basic:1</deviceType>
      <friendlyName>Phoscon-GW (192.168.2.8)</friendlyName>
      <manufacturer>Royal Philips Electronics</manufacturer>
      <manufacturerURL>http://www.dresden-elektronik.de</manufacturerURL>
      <modelDescription>Philips hue compatible Personal Wireless Lighting</modelDescription>
      <modelName>Philips hue bridge 2015</modelName>
      <modelNumber>BSB002</modelNumber>
      <modelURL>http://www.dresden-elektronik.de</modelURL>
      <serialNumber>***censored***</serialNumber>
      <UDN>uuid:***censored***</UDN>
      <presentationURL>index.html</presentationURL>
      <iconList>
         <icon>
            <mimetype>image/png</mimetype>
            <height>48</height>
            <width>48</width>
            <depth>24</depth>
            <url>hue_logo_0.png</url>
         </icon>
      </iconList>
   </device>
</root>

Spitzbua, 2018-11-24:
I can confirm that this issue is fixed with (echo dot v3):

Version 2.05.46
Firmware 26240500

rstoermer, 2018-12-05:
I also can't discover any devices with Alexa, neither by telling Alexa to look for new devices, nor by looking for lamps in the Alexa app. Deconz is running on a RPI3 in a Docker container (https://hub.docker.com/r/marthoc/deconz/) on version 2.05.47. My firmware is 260B0500 (Can't update the firmware to the newest version. In Phoscon I click update, update is running, but after the update is finished, the version is still the old one)

closed 2019-04-11 by stale[bot]

outofsight, 2019-12-26:
Any update on this?
I cannot pair Alexa andoid app with deconz running on RPI 4 HASSIO.

hparusel, 2019-12-26:
Same here. I cannot connect to Alexa either (RPI 1b)...

unstressable, 2019-12-31:
Hey togehter.
For me alexa found my devices connected to phoscon, but there is no possibility to control the lights.
Alexa always says that the devices are unresponsible.
Any idea?

Schnuecks, 2020-03-10:
Same here could not connect to Alexa

rstoermer, 2020-03-25:
Still a problem here...

daubweb, 2020-03-27:
Same issue after a complete reset
Conbee 2 on raspberry pi 3, newest headless sd image.
Not discovered by Alexa...

univok, 2020-03-30:
Conbee 2 on raspberry pi 3, Raspbian Buster with desktop 2020-02-13, stable DeConz from http://phoscon.de/apt/deconz. DeConz and Phoscon work like a charme, thumps up, but Alexa finds not all devices. Some are correct, some missing, and some are found under an older name I changed in phoscon long time ago, and did not work.

Initially, some hue lights and OSRAM/innr smart plugs were able to control by Alexa, but this stopped working all of a sudden, maybe after another "search for devices". Deleting and refinding them doesn't change it. Though Alexa find some devices, no one is responsible, saying "Waiting for Royal Philips Electronics..." IGMP is off at Fritz and Switch. 

Feeling helpless and upset because I have no idea how to debug this. Spent hours of installing, testing. Result is non deterministic. Will be switching to iobroker likely.

hparusel, 2020-03-30:
I really had the same issues. Currently, I use Sonos One, Amazon Flex and one Amazon Echo (3. Generation). Like I already have written before it started working as soon as I got and integrated the Amazon Echo (3. Generation). Last week my wife unplugged the the Echo by coincidence. When I tried to talk to Alexa nothing was responding until I found out that she unplugged the Echo. Even with all Amazon Flex and Sonos One plugged in. As soon as I plugged the Echo in everything was working again. It seems that you need at least one Echo of the current Generation. I know that does not solve you issue. But might get a hint to someone how is able to fix it.

Schnuecks, 2020-03-30:
@hparusel I have only 3rd gen, but still nothing is found. all other apps work with conbee 2 without problems only alexa is currently not functional.

univok, 2020-03-30:
> It seems that you need at least one Echo of the current Generation.
I am using only 3rd generation Echos, a Show 5 and two Dots, as well as the iOS app on the phone. 

I forgot to mention, I didn't flash from the SD Card image avaiable on https://www.phoscon.de/de/conbee2/sdcard, instead installing a clean Raspian Buster and afterwards a "sudo apt install deconz" from "stable main" repository, resulting in deCONZ version 2.05.74 

There is also a "Homebridge-hue" image (beta) at phoscon.de. I left Apple Homekit because I was disappointed, no need for a "Homebridge". But "hue" sounds interessing in this case...

Pirlobeck, 2020-04-12:
I installed a new Fritzbox, and suddenly all lights are not reachable by Alexa and Alexa does not find new lamps. All other things still work find, controling by phoscon or hue Essential.
<root xmlns="urn:schemas-upnp-org:device-1-0">
<specVersion>
<major>1</major>
<minor>0</minor>
</specVersion>
<URLBase>http://192.168.8.1:80/</URLBase>
<device>
<deviceType>urn:schemas-upnp-org:device:Basic:1</deviceType>
<friendlyName>Phoscon-GW-Sven (192.168.8.1)</friendlyName>
<manufacturer>Royal Philips Electronics</manufacturer>
<manufacturerURL>http://www.dresden-elektronik.de</manufacturerURL>
<modelDescription>Philips hue compatible Personal Wireless Lighting</modelDescription>
<modelName>Philips hue bridge 2015</modelName>
<modelNumber>BSB002</modelNumber>
<modelURL>http://www.dresden-elektronik.de</modelURL>
<serialNumber>00212E051FCF</serialNumber>
<UDN>uuid:91be14ad-cc60-41f8-88d4-9e25a2d7855c</UDN>
<presentationURL>index.html</presentationURL>
<iconList>
<icon>
<mimetype>image/png</mimetype>
<height>48</height>
<width>48</width>
<depth>24</depth>
<url>hue_logo_0.png</url>
</icon>
</iconList>
</device>
</root>

Pirlobeck, 2020-04-12:
> I installed a new Fritzbox, and suddenly all lights are not reachable by Alexa and Alexa does not find new lamps. All other things still work find, controling by phoscon or hue Essential.
> 
> 
> 1
> 0
> 
> http://192.168.8.1:80/
> 
> urn:schemas-upnp-org:device:Basic:1
> Phoscon-GW-Sven (192.168.8.1)
> Royal Philips Electronics
> http://www.dresden-elektronik.de
> Philips hue compatible Personal Wireless Lighting
> Philips hue bridge 2015
> BSB002
> http://www.dresden-elektronik.de
> 00212E051FCF
> uuid:91be14ad-cc60-41f8-88d4-9e25a2d7855c
> index.html
> 
> 
> image/png
> 48
> 48
> 24
> hue_logo_0.png

Update
I installed the App for the hue Bridge V1, run this app => it told me, one Bridge V2 found, I should install the new app, but I didn't, but I started a new search with alexa and surprise, all devices are found!

Schnuecks, 2020-04-12:
Tried that app v1 says the same but alexa don't find anything. Is it maybe the problem that the app found v2 but in alexa you can only search v1 without official hue skill?

Pirlobeck, 2020-04-12:
> Tried that app v1 says the same but alexa don't find anything. Is it maybe the problem that the app found v2 but in alexa you can only search v1 without official hue skill?

With alexa I did the general search, not the Alexa V1!

Schnuecks, 2020-04-12:
Still the same with find new devices,nothing.

Pirlobeck, 2020-04-12:
My alexa found only lamps and plugs (also as lamp), no sensors.

univok, 2020-04-12:
Meanwhile, I tested Alexa with deconz again, and also with node-red and the "node-red-contrib-deconz palette".  

Another playground, but same problems:  Alexa device discovery intially works for a while, then more and more eratic and stops finding new devices. Retrying by deleting all devices in the app or at alexa.amazon.de doesn't fix it at all, instead the deleted devices are found agai, sometimes, sometimes not, sometimes never again. 

But then I noticed a very strange detail: alexa even find devices if deconz is not running, so it obviously discovers ghost devices, cached from the cloud, or where only God knows. Bullshit!

What I did now was the following: I removed the Echo Dot 3 from the list of my devices, but not at the alexa.amazon.de menu, instead from https://www.amazon.de/hz/mycd/myx#/home/devices/1. Here you can find a second device list for authorization issues, that seems to trigger a reset of the cloud registration. I even deleted my iOS app, re-signed for the amazon alexa app and a full reset of the Echo Dot 3, as well as re-authorization of the amazon cloud.

The next device discover was successful at first attempt, I can play with alexa again, after 50-100 moronic "Discover my devices", good for nothing. 

I don't know why this works so fuzzy, and it's time wasting to find out. But it definately has nothing to do with deconz or the node-red framework, my fritz router or network confguration. It's alexa.

BenAndDevelopment, 2020-06-14:
I am running phoscon on an Intel nuc on port 80 in docker and can’t get anything connected to Alexa. 
The description.xml looks like this:

`1 0 http://192.168.178.123:80/ urn:schemas-upnp-org:device:Basic:1 Phoscon-GW (192.168.178.123) Royal Philips Electronics http://www.dresden-elektronik.de Philips hue compatible Personal Wireless Lighting Philips hue bridge 2015 BSB002 http://www.dresden-elektronik.de 00212E05A89E uuid:dac43913-e4fd-416a-aefc-291c575a0cec index.html image/png 48 48 24 hue_logo_0.png `

Any idea? I connected everything to a Netgear switch and the echos are connected to a Netgear WiFi.

nin9s, 2020-06-15:
> Meanwhile, I tested Alexa with deconz again, and also with node-red and the "node-red-contrib-deconz palette".
> 
> Another playground, but same problems: Alexa device discovery intially works for a while, then more and more eratic and stops finding new devices. Retrying by deleting all devices in the app or at alexa.amazon.de doesn't fix it, instead the deleted devices are found again, sometimes, sometimes not, sometimes never again, and become unresponsive.
> 
> But then I noticed a very strange detail: Alexa even finds formerly found devices - but when deconz is not running! I couldn't believe it, but it obviously discovers ghost devices, cached from the cloud, or where only God knows. New devices are not found, that's bullshit!
> 
> What I did now was the following: I removed the Echo Dot 3 (and all my other echo devices) from the list of "my devices", but not at the alexa.amazon.de menu, instead from https://www.amazon.de/hz/mycd/myx#/home/devices/1. Here you can find a second device entry for authorization issues, that seems to trigger a reset of the cloud registration if removed. I even deleted my iOS app from the list in the web and from my phone, re-installed it and re-signed for the app, and did a full reset of the Echo Dot 3, as well as re-authorization to the amazon cloud.
> 
> The next device discover was successful at first attempt, I can play with Alexa again, after 50-100 moronic "Discover my devices", good for nothing.
> 
> I don't know why this works so fuzzy, and it's time wasting to find out. But it definately has nothing to do with deconz or the node-red framework, my fritz router or network configuration. It's Alexa.

did you ever found a consistent workaround? I dont want to delete all my devices from my amazon account because it will result in a time consuming process to re-add them all.

Currently my alexa wont find any deconz devices after the deconz ip changed. Have tried to delete them from alexa but it does not made a difference. Any other app can discover the deconz devices  without issues, only alexa cant.

ooii, 2020-06-15:
How many devices are connected to your hub?

nin9s, 2020-06-15:
10, not even close to the limit

ooii, 2020-06-15:
That was my issue, the number of devices. Sorry, not other clue right now.

BenAndDevelopment, 2020-06-18:
Can nobody help with my issue?

Pirlobeck, 2020-06-18:
I'm also in trouble again, I had to change my Raspberry, so phoscon got a new IP, and all lights are no longer reachable by alexa. Phoscon works normal and Hue Essential as well. Alexa says "Server Not recheable" or "Server don't react"

[10 more replies]