dresden-elektronik/deconz-rest-plugin #760: Cannot see single key Aqara switch in Switch management

etique57, 2018-08-29:
Hello,

Using 2.5.35 / pwa 17.08.2018, I can add the Aqara single key wall switch (wireless) (recognized as transmitter 1-gang, added through the Sensors page), but then I cannot see it in the Switch management page.

It works perfectly for 2-gang or single button Aqara switches.

mysteryan, 2018-08-31:
A lot of working switches don’t have a configuration page in the web app yet.  If it’s properly recognized, I suggest you look for the corresponding button event in the websocket or rest api and manually create a rule you like for it.

etique57, 2018-09-01:
Hello,

It's actually the WLXG03LM (the single key zigbee button). The round aqara switch (up to quadruple click) works fine.

SkorpionNL, 2018-09-02:
Same issue here. I can attach an action to the button, but it doesn’t save.

etique57, 2018-11-25:
Hello, any news on this particular switch? The dual-button version works very well, but the single switch is still not supported as of 2.05.48.

I tried to trick the system: mapping a dual button Aqara switch and then changing the sensor uid in the resulting rule but it didn't work. Any clue why?

carloscae, 2018-12-04:
I also have a couple of single rocker Aqara remote switch (old version) that is not configurable on Phoscon app. Doing automations via Hassio, but I guess directly inside Deconz would make system more responsive.

manup, 2018-12-04:
We are working on the Phoscon App Switch editor support for the following switches:

* WXKG01LM
* WXKG11LM
* WXKG12LM
* WXKG02LM
* WXKG03LM

Some of these have different versions wich will also be handled by the Switch Editor.

Current plan is to get a first preview online till Saturday, since I still don't have all switches here for testing there might be issues which need to be ironed out in further releases.

closed 2019-04-10 by stale[bot]