dresden-elektronik/deconz-rest-plugin #763: Set offset to himidity sensor?

FezVrasta, 2018-08-30:
My Aqara humidity sensor is off by 20 points, is there any way to set an offset in DeCONZ to adjust the measurements?

ebaauw, 2018-08-30:
I only created `config.offset` for temperature sensors only, but it could easily be extended to humidity sensors as well.  I do find, however, that I’m not using it, since the deviation varies over temperature.  If I set the offset, so the value shows correctly for 21°C, it’s still off at 15°C.

_With one watch, a man knows what time it is.  With two, he’s never sure._

FezVrasta, 2018-08-30:
If you may add it to the humidity sensors I can easily monitor the values and see if they look off. I have other 2 humidity sensors that work reliably and all of them log to Home Assistant.

vandalon, 2018-09-05:
Can you tell me how to use this config.offset for both? :)

vandalon, 2018-09-05:
Nevermind, figured it out. 

`curl -X PUT -d '{"offset":180}'  http://192.168.1.15/api/<api key>/sensors/16/config`

closed 2019-01-04 by FezVrasta

kugelkopf123, 2019-01-21:
Where can i find this config.offset File?!? Or how can i set it up? i use deconz in Headless mode and the deconz adapter for iobroker. My Aqara Temp.Sensors have a Datapoint Offset but its not writeable.

manup, 2019-01-21:
It can only be set via REST-API. If it is not writeable in ioBroker please reach out to their devs and ask them if they can change it to be writeable.

rammstone83, 2020-11-28:
Hi, i tried to change the Temperature config.offset for an Xiaomi Door Sensor, but unfortunately its not there for the Xiaomi Door Sensors.

SwoopX, 2020-11-29:
That is because those device have no temperature measurement capabilities.

rammstone83, 2020-11-30:
Sorry thats not true, the have an internal Temperature sensor. for example: 
{
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true,
        "temperature": 2600
    },
    "ep": 1,
    "etag": "xxx",
    "lastseen": "2020-11-30T11:51Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.sensor_magnet.aq2",
    "name": "Tür_Gartenhütte",
    "state": {
        "lastupdated": "2020-11-30T11:51:03.320",
        "open": false
    },
    "swversion": "20161128",
    "type": "ZHAOpenClose",
    "uniqueid": "xxx"

but the offset objekt is missing in the config part.

SwoopX, 2020-11-30:
Do you really want to argue with with a guy implementing that stuff and having a sniffer available? 😉 

No, it hasn't. That is supposed to be the device temperature and that is inaccurate as hell. I'd rather prefer to fully strip that away as it is utterly useless.

rammstone83, 2020-11-30:
that could be possible, idk know how accurate it is yes, but also there is a temperature reading, and therefore i was only asking if there is the possibility to define an offset for it. Looks like its not possible at the moment, so my question is answered.

steljwagh, 2023-11-15:
If you want a offset for humidity and or temperature sensors in deconz, it's fairly easy to get a API key from phoscon/help/API information and get a existing key from the whitelist (search for whitelist and you'll find the keys)

Filter on sensors to find the right ID for your sensor.

Then open a command/terminal box and construct the following command to your needs.

`curl -X PUT -d "{\"config\": {\"offset\": 100 } }" http://ipaddress:port/api/key/sensors/10`

offset examples:

-80 = minus 0,8 degrees
100 = plus 1 degree