dresden-elektronik/deconz-rest-plugin #765: Ikea lights flagged as unreachable. Single light control broken, but follows group commands.

olemr, 2018-09-01:
Quite often, one of my 30 Ikea lights are flagged as unreachable.
Some times the matter resolves itself, sometimes not.
The light on the picture below has been flagged unreachable for several days.
Still follows group commands, but does not respond to single light commands.
Running the beta 2.05.35 version. (Headless on ubuntu)
I have 10-15 Ikea lights/panels still on the Ikea Gateway and am a bit reluctant to go 'all-in' on Conbee/deconz because of this issue.

Cheers.

[image]

olemr, 2018-09-10:
Probably duplicate of #771.
No devices offline for 3 days ...

manup, 2018-09-10:
Cool, let's see how it holds, fingers crossed.

olemr, 2018-09-11:
One light went offline last evening.
I'm running with debug on. Anything I could look for in the log?

olemr, 2018-09-14:
I power-cycled my 16 outdoor lights, and 24hrs later the same light became unreachable.
This time it did not respond to group commands so it could be a bad bulb.
Will try to swap it with another.

olemr, 2018-09-19:
It happened again.
Another bulb in the exact same spot became unreachable after two days.
Strange, since this is an outdoor lamp with two bulbs in it mounted just a few cm from each other.
The twin bulb is always reachable.

manup, 2018-09-19:
It's hard to tell what exactly is going on, a sniffer log would be very helpful.

Are your lights updated to the latest IKEA firmware version?

SkorpionNL, 2018-09-19:
Also have this issue where individual lights are flagged as unreachable but respond to the group command. This affects both Tradfri as Hue.

olemr, 2018-09-20:
Mine does not respond to group or Ikea remote commands.
Will check bulb Firmware versions tonight.
But things have definitely improved after the latest Conbee firmware.
Only this one bulb acting up now.

manup, 2018-09-23:
> Does the sniffing need to be done with an extra Conbee?

Yes, any other sniffer will do also. So it's not possible to run ConBee as gateway and sniffer at the same time.

ErnstEeldert, 2018-09-24:
I have the same issue with a IKEA GU10 white bulb and a 30W driver. Resetting the device doesn't make it respond though.

manup, 2018-09-25:
>I have the same issue with a IKEA GU10 white bulb and a 30W driver. Resetting the device doesn't make it respond though.

>Devices do respond do connected dimmer and switch.

Can you provide more details, which firmware versions to the devices and RaspBee/ConBee have. Screenshot of the network would also be useful.

ErnstEeldert, 2018-10-09:
The lights required a power cycle to get back in working order again. For the driver, I removed it from the network through PWA, then did a power cycle and reset. It's working again now. This was on 2.0.38 and fw 26250500. As I have upgraded (deconz) and downgraded (firmware) in the mean time, not possible to make a network screenshot.

Is there a way of extracting the network state other than making screenshots of specific devices/clusters? I run deconz in headless mode, so acquiring this info requires me to stop deconz, start deconz-gui, connect to vnc, locate the device (hard in a 60+ node network), make the screenshot, restart deconz etc. Perhaps a simple command line interface with ability to filter by mac address?

closed 2019-02-13 by stale[bot]