dresden-elektronik/deconz-rest-plugin #766: Pairing fails after some network uptime unless reconnecting the dongle

ausserlesh, 2018-09-02:
After some operation time, deCONZ (rev. 2.05.20) with ConBee dongle (firmware versoin 0x261F0500) does not allow pairing of new devices or previously paired, but then lost devices #518.

Packet analysis revealed that in the bad case the ZigBee beacon frame looks as follows

```
ZigBee Beacon, ZigBee PRO, EPID: Dresden-_ff:ff:00:xx:xx
    Protocol ID: 0
    Beacon: Stack Profile: ZigBee PRO, Router Capacity
        .... .... .... 0010 = Stack Profile: ZigBee PRO (0x2)
        .... .... 0010 .... = Protocol Version: 2
        .... .1.. .... .... = Router Capacity: True
        .000 0... .... .... = Device Depth: 0
        0... .... .... .... = End Device Capacity: False
    Extended PAN ID: Dresden-_ff:ff:00:xx:xx (00:21:2e:ff:ff:00:xx:xx)
    Tx Offset: 16777215
    Update ID: 6
```

whereas in the good case the beacon also reports ``End Device Capacity=true``.

```
ZigBee Beacon, ZigBee PRO, EPID: Dresden-_ff:ff:00:b1:09
    Protocol ID: 0
    Beacon: Stack Profile: ZigBee PRO, Router Capacity, End Device Capacity
        .... .... .... 0010 = Stack Profile: ZigBee PRO (0x2)
        .... .... 0010 .... = Protocol Version: 2
        .... .1.. .... .... = Router Capacity: True
        .000 0... .... .... = Device Depth: 0
        1... .... .... .... = End Device Capacity: True
    Extended PAN ID: Dresden-_ff:ff:00:b1:09 (00:21:2e:ff:ff:00:b1:09)
    Tx Offset: 16777215
    Update ID: 6
```

The (Xiaomi) end device ignores the beacon message in the negative case and not send an association request.

12 end devices are connected to the dongle in the bad case. This should not be a problem - when pairing all devices right after connecting the dongle and starting deCONZ I already had 20+ end devices in the network.

/cc @ebaauw

ebaauw, 2018-09-02:
There's a limit to the number of end devices _per parent_ (a router, incl. the coordinator), I seem to remember this limit is 10 for the ConBee/RaspBee on recent firmware.  The _network_ can connect as many end devices as the routers can accommodate.

The parent device needs to cache requests for the end-device until the end-device wakes up, hence the limit.  I think in your "bad" case, the ConBee is at its limit and simply cannot accept any more end devices.  Reconnecting the ConBee is like power cycling it: it clears the list of end devices currently connected so the ConBee is ready to accept new ones.  The old end devices should find a new parent when they wake up, as they've already joined the network.

I'm not too familiar with the lower layers in the ZigBee stack, but I think other routers should also send beacons.  The end device should pick up one from a router that still has room.

Tip: You can also use `sudo GCFFlasher_internal -r -d /dev/USB0` to reset the ConBee without having to remove and reinsert it.

ausserlesh, 2018-09-02:
Thank you for the clarification, in this case the observation is not related to your original issue.

closed 2018-09-02 by ausserlesh