dresden-elektronik/deconz-rest-plugin #767: all nodes lost overnight

ErnstEeldert, 2018-09-03:
Hey manup,

Here's a new one I hadn't encountered yet. A couple of days ago, lights stopped responding to commands. As I didn't have time to dig into it, I used the wall switches.

Today, I had a look, and it looked like I've lost all lights in PWA. So I restarted deconz, same result. Upgraded from .31 to .37, lights (and sensors and switches I think) got back, but all in unavailable state. Upgraded firmare from e500 to the latest, same result.

As I run deconz in headless mode, I decided to run it in headed mode, and the network diagram shows me the raspbee, but nothing else. Here's the node info:

[image]

What should I do to rebuild the zigbee mes(s|h), without loosing all my config on the deconz (and related homebridge) side of things? I have around 50 lights here...

Thanks!

manup, 2018-09-03:
I guess the networks settings are messed up, likely the channel.

The *zll.db* table `zbconf` contains all network setting changes, a bit hacky but I would suggest opening the file with sqlitebrowser and look if the channel has changed (or other parameters).

ErnstEeldert, 2018-09-03:
Channel seems to be fine, has been stuck on 25 since the start. However, the latest change is showing a different network key. You can see the full contents of the table [here](https://gist.github.com/ErnstEeldert/146fc29ef2bce3eb55a1659b5d502ae9).

manup, 2018-09-03:
Ups did reply in the gist :)

ErnstEeldert, 2018-09-04:
Setting the network key and panid as per the old values and rejoining the fixed the issue. All nodes are now returning to deconz.

Scary stuff though that this kind of corruption just happens overnight :(

closed 2018-09-04 by ErnstEeldert