dresden-elektronik/deconz-rest-plugin #768: Completely Reset ConBee & deCONZ?

felixstorm, 2018-09-03:
I just tried for several hours to get a ConBee stick to work (primarily tested with deCONZ 2.05.35 on Windows and firmware 0x26210500, but also with deCONZ 2.05.35 and 2.05.37 in WSL and also with firmware 0x26220500), but for whatever reason I am unable to join any device (hue bulb, Aqara sensor, ubisys C4) to the network.

What I have tried so far:
* Click "Reset gateway" in Phoscon App and restart deCONZ.
* Quit deCONZ, delete deCONZ folder with database and start deCONZ again
* Manually configured network settings in the deCONZ UI
* Unplugged ConBee several times

The ConBee is being displayed as the only device in deCONZ as always, but other devices seem to be unable to join (I opened the network through the deCONZ UI for 60 seconds, 100 seconds and forever and I also tried to search for lights through the Phoscon app). The ConBee definitely used to work (I already used it some weeks ago for some testing in addition to the RaspBee working in production). To make sure the hardware is ok, I also used it for sniffing with zshark just now which did work without problems.

Any hint to completely reset ConBee (i.e. NVRAM with network parameters) & deCONZ for a clean start or other ideas what can possibly still be wrong here? Being pretty frustrated now, anything is welcome ;-)

Thanks,
Felix

manup, 2018-09-03:
> Manually configured network settings in the deCONZ UI

Can you provide a screenshot of the settings, some parameters are very important to let other nodes join the network.

felixstorm, 2018-09-03:
Thanks for your quick reply @manup - these are my current values:
[image]

manup, 2018-09-03:
The settings look correct, TC Link Key must be 5a6967426565416c6c69616e63653039.

Since you use a ConBee you may try to use a extension cable to get some space between ConBee and PC, we've seen very poor signal in some setups when the ConBee is connected directly to the PC.

felixstorm, 2018-09-03:
You're a genius, @manup :-)

I had the ConBee plugged into an USB hub (which I am pretty sure did use to work and still worked for sniffing earlier today), but connecting it to the computer's direct USB port made it work completely and I can now join other devices again.

Thanks a lot,
Felix

closed 2018-09-03 by felixstorm

ronnyandre, 2020-01-04:
I need to completely reset my ConBee and deCONZ, any ideas?

mdolnik, 2020-02-09:
If anyone wants to completely reset deCONZ on linux (Ubuntu) they can do the following...

(_**Disclaimer: I figured this out through trial and error so there may be issues with this approach**_)

- Uninstall deCONZ dev version: `sudo apt remove deconz-dev -y`
- Uninstall deCONZ: `sudo apt remove deconz-dev`
- Remove common files: `sudo rm -rf /usr/share/deCONZ/`
- Remove user files: `sudo rm -rf ~/.local/share/dresden-elektronik`
- Reinstall deCONZ dev version: `sudo apt install deconz-dev -y`

Or do all of this in one command:
```
sudo apt remove deconz-dev -y && sudo apt remove deconz -y && rm -rf /usr/share/deCONZ/ && sudo rm -rf ~/.local/share/dresden-elektronik && sudo apt install deconz-dev -y
```

This will destroy all configuration and connection to devices and bring deCONZ back to a freshly installed version.

I had to do this a few times when I was messing with the code and added devices to the DB that I could not remove through the UI, resetting everything helped me to debug things.

andras-tim, 2020-02-17:
> This will destroy all configuration and connection to devices and bring deCONZ back to a freshly installed version.

Yes, but how can I reset the dongle? E.g. if you add a new node to a fresh dongle, the internal ID will be 1 (check the debug logs of deconz), but if I try things after your "reset" method, the ID will be n+1 where the n is the last used ID before the "reset".