dresden-elektronik/deconz-rest-plugin #769: Changes required for ssl support

KodeCR, 2018-09-03:
As mentioned in #708 and #675 the latest updates of the Hue App require an https connection to the bridge with a self-signed certificate. Are there any plans to support this?

It looks like deconz has a built-in web-server based on Qhttp. I can't really help out any further since that part is closed-source, but based on https://www.thomaskeller.biz/blog/2009/01/03/ssl-verification-with-qt-and-a-custom-ca-certificate/ it is very easy to add SSL support with a specific certificate:

```
QHttp * con = new QHttp();
QFile certFile("path/to/root.crt");
Q_ASSERT(certFile.open(QIODevice::ReadOnly));
QSslCertificate cert(&certFile, QSsl::Pem);
// this replaces the internal QTcpSocket QHttp uses; unfortunately
// we cannot reuse that one because Qt does not provide an accessor
// for it
QSslSocket * sslSocket = new QSslSocket(this);
sslSocket->addCaCertificate(cert);
httpConnection->setSocket(sslSocket);
con->setHost("some.host.com", QHttp::ConnectionModeHttps);
con->get("/index.html");
// connect to QHttp's done() signal and read the response

```

From https://github.com/mariusmotea/diyHue/commit/b96603af630f340ebdcf432c6c2e56ee42852b3b there may be some other SSL settings required and to be passed to QSslConfiguration to make the Hue App happy. Maybe @mariusmotea can share what is required there?

Thanks!

mariusmotea, 2018-09-03:
I believe the big challenge is to generate the certificate than to have proper ssl support.

For https server these are ssl parameters that need to be implemented:

```
       ctx.options |= ssl.OP_NO_TLSv1
        ctx.options |= ssl.OP_NO_TLSv1_1
        ctx.options |= ssl.OP_CIPHER_SERVER_PREFERENCE
        ctx.set_ciphers('ECDHE-ECDSA-AES128-GCM-SHA256')
        ctx.set_ecdh_curve('prime256v1')
```
For certificate generation look my easy_install.sh script, with current version i don't have complains.

manup, 2018-09-03:
The deCONZ sources contain a HTTPS server for some years now but it's not enabled and likely won't compile today, it will return at some time but there is no ETA yet.

KodeCR, 2018-09-30:
In case anyone is interested, I created a pass-through https server for deconz based on bits and pieces from diyHue. It automatically creates the certificate on the first run. You can find it here: https://github.com/KodeCR/hue-pass

closed 2019-02-05 by stale[bot]